webcamoid/webcamoid #263: FFMpeg sink crashes as AV_CODEC_ID_AMR_NB not supported

illuusio, 2020-04-04:
#Summary

As FFmpeg used 3G2 default output and AMR NB is not supported (in openSUSE) at least it crashes
https://github.com/webcamoid/webcamoid/blob/84c42c528932ac9d7b5918f54c153ef9e137c0e5/libAvKys/Plugins/MultiSink/src/ffmpeg/src/mediawriterffmpeg.cpp#L320 when it tries to get name from NULL pointer.

## Current Behavior

Webcamoid doesn't start without removing FFmpeg sink.

## Expected Behavior

Webcamoid not to sigsegv on start

## Steps to Reproduce

Install openSUSE with default FFmpeg 4.x and compile Webcamoid

## Your Environment

* Webcamoid information: GIT
* Operating System information: openSUSE Tumbleweed

hipersayanX, 2020-04-04:
It may be fixed in the last commit.

illuusio, 2020-04-06:
Seems to start normally. I think it's fixed.  I have to learn more to see is it really working but closing. Thanks.

closed 2020-04-06 by illuusio