webcamoid/webcamoid #264: VirtualCameraElement can crash because memcpy target can be smaller than source

demiantres, 2020-04-04:
VirtualCameraElement can crash because  frame.data().size() can be smaller than videoPacket.buffer().size(). The reason for this could be that the videoPacket has aligned bytes (the number of bytes in a row is bigger than the actual frame's width).

```
#include <QDebug>
AkPacket VirtualCameraElement::iVideoStream(const AkVideoPacket &packet)
{
    this->d->m_mutex.lock();

    if (this->state() == AkElement::ElementStatePlaying) {
        auto videoPacket = packet.convert(AkVideoCaps::Format_rgb24, 32);
        auto fps = AkVCam::Fraction {uint32_t(videoPacket.caps().fps().num()),
                                     uint32_t(videoPacket.caps().fps().den())};
        AkVCam::VideoFormat format(videoPacket.caps().fourCC(),
                                   videoPacket.caps().width(),
                                   videoPacket.caps().height(),
                                   {fps});
        AkVCam::VideoFrame frame(format);
        if (frame.data().data() && videoPacket.buffer().constData() && videoPacket.buffer().size() > 0)
        {
            //CRASH: frame.data().size() can be smaller than videoPacket.buffer().size()
            qDebug() << "Size: " << frame.data().size() << videoPacket.buffer().size();
            /* memcpy(frame.data().data(),
                   videoPacket.buffer().constData(),
                   size_t(videoPacket.buffer().size())); */
            this->d->m_ipcBridge.write(this->d->m_curDevice.toStdString(), frame);
        }
    }

    this->d->m_mutex.unlock();

    akSend(packet)
}
```

demiantres, 2020-04-05:
The following bugs are very likely also related to this:
https://github.com/webcamoid/webcamoid/issues/253
https://github.com/webcamoid/webcamoid/issues/252

Eventually also this one:
https://github.com/webcamoid/webcamoid/issues/245

demiantres, 2020-04-05:
A quick and dirty fix is this:

```
AkVCam::VideoFrame frame(format);
if (frame.data().size() == size_t(videoPacket.buffer().size()))
{
	memcpy(frame.data().data(),
	   videoPacket.buffer().constData(),
	   size_t(videoPacket.buffer().size()));
}
else
{
	int bytesPerRow = format.width() * (format.bpp()/8);
	int totBytes = format.height() * bytesPerRow;
	int stride = bytesPerRow + ((videoPacket.buffer().size()-totBytes) / format.height());

	auto *ptrDst = frame.data().data();
	auto *ptrSrc = videoPacket.buffer().constData();
	for (int i=0; i<format.height(); i++)
	{
		memcpy(ptrDst, ptrSrc, bytesPerRow);
		ptrDst += bytesPerRow;
		ptrSrc += stride;
	}
}
this->d->m_ipcBridge.write(this->d->m_curDevice.toStdString(), frame);
```

hipersayanX, 2020-04-07:
That fix will need even more work, many things in [VCamUtils/image](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/VCamUtils/src/image) will need some redesign, specially VideoFormat and VideoFrame.

demiantres, 2020-04-07:
I see. At least the dirty fix from above seems to prevent webcamoid from crashing.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX