webcamoid/webcamoid #265: How to create packages

demiantres, 2020-04-05:
I followed the instructions from https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows. The result is only a single zip file "webcamoid-portable-8.7.1-win64.zip" (win64 is added although it's a 32 bit build). The zip file contains some files (D3DCompiler_47.dll, libgcc_s_dw2-1.dll, webcamoid.bat, bin\, share\).

How can I create a package which includes all plugins, and the exe files?

demiantres, 2020-04-06:
I also tried msys2 on a fresh machine

1. install_reps.sh
2. build.sh
3. deploy.sh

The result is again a zip file containing only the Qt libraries but no webcamoid.exe nor any of the plugins.

hipersayanX, 2020-04-07:
In Qt Creator go to _Projects > Build Steps > Add Build Step > Custom Process Step_ and set

**Command**: python3
**Arguments**: ports/deploy/deploy.py

If you have enabled the shadow build (in the **General** section, in the top), then go to **Build Environment**, and add the BUILD_PATH variable with the same path as the shadow build.
Just remember to disable this step if you are just debugging Webcamoid because this step is very slow.
For creating the installer you must install the [QtIFW](https://download.qt.io/official_releases/qt-installer-framework/).

>I also tried msys2 on a fresh machine
>
>    install_reps.sh
>    build.sh
>    deploy.sh

Those scripts are only for AppVeyor, Travis and Circle, they are not supposed to be run by users.

demiantres, 2020-04-07:
Thanks @hipersayanX . Interestingly the package created this way contains only the Qt libraries but not the webcamoid dlls or exe. As a workaround I just copied all files manually (using a bat file).

hipersayanX, 2020-04-07:
Could it be that you compiled as Debug? because I've only tested the deploy scripts to work with the Release version.

demiantres, 2020-04-07:
I compiled in Release mode?

hipersayanX, 2020-04-07:
The script is working fine according to [this](https://ci.appveyor.com/project/hipersayanX/webcamoid/build/job/awd3xxffww7igin7#L3098). If you had setup everything right it may be working.

closed 2021-02-16 by hipersayanX