webcamoid/webcamoid #266: App crashes when video source is set

demiantres, 2020-04-06:
The app crashes when a video source (an url or a video file) is set as video source with the latest Master branch (x86 release build) on Windows.

Steps to reproduce:

1. Start webcamoid and set a video file as video source.
2. Exit the webcamoid app.
3. Start the app again. The app shows the first frame from the video file from step 1. but immediately crashes.

obszczany, 2020-04-06:
Hello, could you please explain it more, like how do i do it Exactly?
my app dont even wants to start now, its just crashes at the same time.

hipersayanX, 2020-04-07:
@obszczany https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh

demiantres, 2020-04-07:
It could be that https://github.com/webcamoid/webcamoid/issues/264 fixes the problem. At least I cannot reproduce the problem anymore with the patch from https://github.com/webcamoid/webcamoid/issues/264.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX