webcamoid/webcamoid #267: Donating to the project

fangfufu, 2020-04-06:
Considering how useful and fun this project is, I would like to donate to the project. Is there a link for donation?

hipersayanX, 2020-04-07:
Not for now, unless I've the guarantee that I'll receive at least the minimum money to pay for taxes and keep a bank account up, **EVERY MONTH,** I won't receive donations. The best I can do for now would be receive Bitcoins if there are people interested on it.

fangfufu, 2020-04-07:
Surely you can just get people to send money to one of your personal account, and pay income tax on it? Although I have no idea which jurisdiction you are in, and your local tax regulation.

hipersayanX, 2020-04-07:
>Surely you can just get people to send money to one of your personal account, and pay income tax on it?

Yes but I need a minimum money guaranteed every month, otherwise I will just loosing money paying taxes and BA from my own savings, donations does not works since it's sporadic, I've checked other projects with similar numbers and similar popularity that works with donations, and my numbers needs to be a lot much bigger.

>Although I have no idea which jurisdiction you are in, and your local tax regulation.

:argentina: (aka :venezuela: 2.0).

fangfufu, 2020-04-07:
Ah, as long as you have looked into it. :) 

I am from the UK / England. If we have additional income from outside our job, we just fill in a Self Assessment Tax Returns. If I had a project like yours, I would just leave the email address in the public, and get people to send me money via Paypal.

closed 2020-04-07 by fangfufu

hipersayanX, 2020-04-07:
Don't worry, I'm just waiting a bit until things get better :wink: