webcamoid/webcamoid #269: Virtual Camera Not Working

Happy225, 2020-04-08:
Hi!
I am very interested in your repository, so I have just build webcamoid on windows 10 by myself.
I used VisualStudio2017 , Qt5.12 and ffmpeg-20200403. 
Built as 64bit.
I succeeded to build application. Very very appreciate of it. :)
As next step, I added a VIRTUAL CAMERA.
Google Chrome detected this virtual camera successfully.
But Skype and Discord doesn't detect this one.

I hope to know what is the problem. and what's wrong on my operation.

Also I checked your prebuilt exe and installers from your repository.
As starting it, it is very very slow. I think this is critical issue.
After some minutes, it opened .
But , When I try to add a virtual camera, it was crashed.

I want to know the reason of this.

Happy225, 2020-04-08:
I have just rebuilt as 32 bit.
But virtual camera doesn't work.
Used VS2017 and Qt5.14 for 32bit

hipersayanX, 2020-04-12:
Check #257.

closed 2020-04-12 by hipersayanX