webcamoid/webcamoid #273: Getting instructions to "enable virtual webcam output"?

jasonsemko, 2020-04-10:
[image]

[image]

Have googled and searched these issues and no where can I find how to:

- Go to Webcamoid (the app?)
- Enable virtual webcam output (Google for these instructions and didn't find anything. Where is this supposed to be located?)

Zoom only sees Facetime camera, not virtual ones.

Are there any instructions for how to use this app? Wiki just shows what is compatible.

hipersayanX, 2020-04-12:
Check [this](https://github.com/webcamoid/webcamoid/issues/220#issuecomment-599091191).

jasonappah, 2020-04-22:
I believe Zoom recently patched something in 1.6.9 and higher that stopped virtual cameras from working. I downgraded to 1.6.8 and got a different virtual camera program to run so maybe you could try that.

gh-markt, 2020-11-08:
I am experiencing what I believe may be a related issue.  In my case, the virtual webcam is selectable by other apps, but only produces the single static image which the OP above had shown, directing to enable virtual webcam output    I have not been able to find any way  enable the webcam output  in the webcamoid app to direct the source to the virtaul webcam output, however.  I am using linux with akvcam as a virtual camera driver.  I experience the same effects with zoom as well as in-browser apps which access the webcam whenever I set the webcam device to the virtual one, and do not experience any issues using the regular webcam.

hipersayanX, 2020-11-09:
@gh-markt Hi, this issue is specific for Mac, you must open one for GNU/Linux (or check if one already exists), also for using the virtual camera is as simple as just creating a camera, selecting it, and then play your webcam, your desktop or a video, if it does not works then we are talking about a bug.

closed 2021-02-16 by hipersayanX