webcamoid/webcamoid #275: a crash after removing usb wlan adapter

beppe9000, 2020-04-11:
i have a dump. maybe it's useful, maybe not. or maybe this crash is overly specific to my environment.
[webcamoid.exe.44640_wifi_key_unplugged.dmp.zip](https://github.com/webcamoid/webcamoid/files/4464991/webcamoid.exe.44640_wifi_key_unplugged.dmp.zip)

hipersayanX, 2020-04-12:
That bug does not looks reproducible, and could be pure coincidence. Anyway, it's possible to convert that .dmp file to some readable format like plain txt or html, so I can check from GNU/Linux?

beppe9000, 2020-04-12:
I have no idea... they are created automatically by windows. I don't even know how one is supposed to use them for debugging. I suppose it may boil down to a stacktrace somehow

beppe9000, 2020-04-12:
it seems windbg is used

```

Microsoft (R) Windows Debugger Version 10.0.10586.567 X86
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\Administrator\AppData\Local\CrashDumps\webcamoid.exe.44640_wifi_key_unplugged.dmp]
User Mini Dump File: Only registers, stack and portions of memory are available

************* Symbol Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       symsrv*symsrv.dll*C:\Windows\Symbols*http://msdl.microsoft.com/download/symbols
Symbol search path is: symsrv*symsrv.dll*C:\Windows\Symbols*http://msdl.microsoft.com/download/symbols
Executable search path is: 
Windows 10 Version 14393 MP (8 procs) Free x86 compatible
Product: Server, suite: TerminalServer SingleUserTS
Built by: 10.0.14393.2430 (rs1_release_inmarket_aim.180806-1810)
Machine Name:
Debug session time: Fri Apr 10 14:51:12.000 2020 (UTC + 2:00)
System Uptime: not available
Process Uptime: 0 days 1:09:56.000
................................................................
................................................................
................................................................
.......................................................
Loading unloaded module list
................................................................
This dump file has an exception of interest stored in it.
The stored exception information can be accessed via .ecxr.
(ae60.62b8): Access violation - code c0000005 (first/second chance not available)
eax=00000000 ebx=00000000 ecx=77157834 edx=00000000 esi=00000003 edi=00000003
eip=7723f29c esp=159cde28 ebp=159cdfb8 iopl=0         nv up ei pl nz na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00000206
ntdll!NtWaitForMultipleObjects+0xc:
7723f29c c21400          ret     14h
0:025> .ecxr
eax=00000000 ebx=12137020 ecx=77157834 edx=00000000 esi=02380000 edi=00000000
eip=77216c61 esp=159ceb28 ebp=159ceb6c iopl=0         nv up ei pl zr na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00010246
ntdll!RtlFreeHeap+0x81:
77216c61 807bff05        cmp     byte ptr [ebx-1],5         ds:002b:1213701f=??

```

I think symbols are needed to show the source code of the exception

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX