webcamoid/webcamoid #277: effect chaining / layering ?

beppe9000, 2020-04-11:
is there any plan to allow that ?

hipersayanX, 2020-04-12:
It's actually possible but very unintuitive, as it's implemented now, you must enable the advanced effect mode and then you can add as many effects as you want, but again is unusable as it's implemented.
The redesign is already planned, check [this](https://github.com/webcamoid/webcamoid/issues/216#issuecomment-589687742) (Video effect selection), the feature that you want wiil be implemented as the **Overlap** option (or some similar name).
I'll start working on that probably this week.

beppe9000, 2020-04-12:
looks promising. indeed I was picturing it like the level panel of photoshop

hipersayanX, 2020-04-12:
Do you mean something like this?

[image]

My idea is to start with something simple and then start adding the rest of the features later, based on what users needs.
Also, I'm very careful not adding features that consumes a lot of resources, for example, I wanted to use a Cheese like effect selection like this:

[image]

but it does not looks like a good idea doing that in a phone.

beppe9000, 2020-04-12:
yes, actually i meant much simpler than that. basically a listbox control with buttons or drag&drop for setting order

hamoid, 2020-04-18:
Would it be possible to add user GLSL effects? It would be cool if that was possible without recompiling.

hipersayanX, 2020-04-19:
@hamoid possible yes, but you must write a plugin that make use of them.

hamoid, 2020-04-20:
Do you mean 1 plugin, maybe called "user effects" that provides a way to load effects? (either a textarea or a dropdown showing effects from a folder), and maybe with a way to build a GUI for those shaders? Using ISF might be a way to get a few thousand effects at once: https://editor.isf.video/shaders?q=&category=&sort=&page=0

hipersayanX, 2020-04-24:
[This feature](https://github.com/webcamoid/webcamoid/issues/216#issuecomment-618757525) can be tested in the daily build. I was not able to allow the user to rearrange the effects by dragging them, the ScrollView seems to be eating the drag & drop events, must figure out how to fix that.

@hamoid Open a new issue so I can remember to do that later.

hamoid, 2020-04-24:
@hipersayanX The new issue: https://github.com/webcamoid/webcamoid/issues/288

hipersayanX, 2022-02-15:
Feature added in 9.0.0, closing.

closed 2022-02-15 by hipersayanX