webcamoid/webcamoid #280: App not working anymore

lostanco, 2020-04-15:
Hello, I'm a simple user not a coder. I installed the app some days ago and it worked for 3 or 4 times but now the app don't work anymore and I have to force quit.

hipersayanX, 2020-04-16:
[Reset configs and starting fresh](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh)

lostanco, 2020-04-16:
Perfect! Thank you so much for your time and effort

lostanco, 2020-04-16:
Sorry...it work one time, then it starts to crash again even if i delete again the plist file

I'm trying to send to webcamoid a stream from my gopro with ffmpeg and camerasuite app...it worked, I saw the video from my gopro once adding udp://localhost:10000 as a device, but only once...when i created a virtual webcam to send the stream to zoom it crashed

hipersayanX, 2020-04-16:
>udp://localhost:10000

That's the problem, Webcamoid keeps waiting for the stream to send some data, and it get stuck, it was already solved in the master a few days ago (#259), instead of resetting the entire program, you can astream with your GoPro, start Webcamoid and then remove the stream.

lostanco, 2020-04-18:
Yes, the problem is that the app won't start again even if I delete all the references I can find on my hard drive including the plist file

hipersayanX, 2020-04-19:
Can you try the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/untagged-d3a4aa8a9bb5aec0c4c5) and confirm if you still have the same problem?

lostanco, 2020-04-21:
It doesn't work with Mac Os 10.12...why did you switch to 10.13? :(

lostanco, 2020-04-24:
Hello...

hipersayanX, 2020-04-24:
@lostanco Don't remember, probably because homebrew was failing with xcode9 or something.

lostanco, 2020-04-24:
is there any chance to drop some lines of code and make it compatible with older OS?

hipersayanX, 2020-04-25:
I've found the log.
[Here](https://docs.travis-ci.com/user/reference/osx/#macos-version) you can check which Xcode version is associated to which macOS version, in this case you want support for macOS 10.12 in which the latest supported Xcode version was xcode9.2 as you can see in [this](https://travis-ci.org/github/webcamoid/webcamoid/builds/593764454) and [this](https://travis-ci.org/github/webcamoid/webcamoid/jobs/593764479#L4390) log, it fails because it needs to build node and other packages from scratch because homebrew does not supports that Xcode version anymore, the build stuck for 10 minutes and fail, support was removed 7 months ago.

>is there any chance to drop some lines of code and make it compatible with older OS?

You can still compile from sources, but due to that problem I can't provide binaries for that macOS version anymore.

closed 2021-02-16 by hipersayanX