webcamoid/webcamoid #281: No camera or media images visible on master

magicseth, 2020-04-16:
# Reporting a problem?

## Summary

Compiling master on Mac (537570f1652222f1a1ed499a3f17e92c680ff944)

I can't see any images from cameras or media.

All the UI looks good, but no camera feed or media comes through.

## Current Behavior

I can compile the app, open the dmg, run the app.
[image]

I've selected a source of my facetime camera:
[image]

I've turned on play sources
[image]

But there's no video showing up. it's all gray

## Expected Behavior
When I run the the app I expect to see my video pass through from my webcam

## Steps to Reproduce

1. I installed dependencies by doing this: export TRAVIS_OS_NAME=osx and running install_deps
2. I configured QT according to the guide
3. I compiled and ran the app

## Suggestions and tips

I'm open to trying things if you want to point me towards logs, or areas to try to fix

## Your Environment

* Webcamoid information: master

Operating System information: (name) (numeric version and codename if applied) (architecture)
Mojave 10.14.6
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

hipersayanX, 2020-04-16:
This is a known bug, it needs [NSCameraUsageDescription](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/Info.plist#L25) which is not available since it does not create an app bundle on build time .

magicseth, 2020-04-18:
Okay.

I just checked out master, and still no dice. I used the deployed dmg.  The green light of the camera comes on, but I still don't see any images.  If there's something you want me to try to investigate, I'm happy to dig in.

Thanks for this awesome work.

hipersayanX, 2020-04-19:
Beside tracing the [source code](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/avfoundation/src/captureavfoundation.mm#L289) if you know C++/Qt, I can't give you much tips here, I'll check when I've some time.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX