webcamoid/webcamoid #286: Streaming a video from C# to AkvCam on windows

beppe9000, 2020-04-21:
What headers should I use with [this tool](https://github.com/mono/CppSharp) ?

hipersayanX, 2020-04-24:
[API header](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/ipcbridge.h)
[API lib](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/VCamIPC)
[API example](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp#L64)

beppe9000, 2020-04-24:
Thanks! I will post when I get it working or hit some roadblock.

hipersayanX, 2020-04-25:
There are many users asking the same, a solution would be designing a communication protocol with the plugin so developers don't need to link against any library to talk to it, maybe using something similar to Linux device drivers in which you communicate with device files. Probably using [File Mapping](https://docs.microsoft.com/en-us/windows/win32/ipc/interprocess-communications#using-a-file-mapping-for-ipc) or something similar.

correa, 2020-06-07:
Would webcamoid accept pull requests and setup a new repository to separate the virtual camera in it's own repository/library to accommodate those users?

hipersayanX, 2020-06-08:
@correa I'm already into that.

https://github.com/webcamoid/akvirtualcamera

hipersayanX, 2021-02-16:
Check latest updates for [akvirtualcamera](https://github.com/webcamoid/akvirtualcamera/issues/3), its almost working fine and no linking is required, so it can be used easily in any programming language.

closed 2021-02-16 by hipersayanX