webcamoid/webcamoid #287: I see no IP setup for Webcamoid

BillyRe, 2020-04-22:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

BillyRe, 2020-04-22:
I see no IP setup for Webcamoid. Is Webcamoid used for IP cams? If not what is it used for?
Thanks

hipersayanX, 2020-04-24:
There's no specific setup for IP cameras, but you can add the video stream of the camera in the Add Media/Source dialog.

BillyRe, 2020-04-24:
Hi, hipersayanX,
Are there steps in a full setup for IP cams?
Thanks

hipersayanX, 2020-04-25:
Just put the [url of the stream](https://www.ispyconnect.com/man.aspx?n=MJPEG) in the path and it may work, or check in your camera manual how to get the url.

XavierIPcam, 2020-05-04:
> Just put the [url of the stream](https://www.ispyconnect.com/man.aspx?n=MJPEG) in the path and it may work, or check in your camera manual how to get the url, or search the vendor and model of the camera in internet to get that information.

Hi @hipersayanX ,

I have the same issue.
I put the url of the stream in the path of the Add Media/Source dialog and, I not get the video.

Any idea ?

Thanks, see you.

hipersayanX, 2020-05-05:
@XavierIPcam what was the url of the stream?
You can go for example to https://www.insecam.org/ and get any stream, for example http://185.39.101.26/mjpg/video.mjpg (this stream is valid at the moment of writing this), and put it in Webcamoid and it will work. The same is valid for any IP camera, but you must know which is the exact URL of the stream.

qnxdev, 2020-07-04:
It's not responding when i add this

hipersayanX, 2022-02-15:
Added _Video > Add source > URL_ option in Webcamoid 9.0.0.

closed 2022-02-15 by hipersayanX