webcamoid/webcamoid #289: USB webcam is not detected and can not be added manually

WildPenquin, 2020-04-25:
Version: 8.7.1
OS: Arch Linux (installed from AUR)

## Summary

Webcamoid does not detect my USB webcam, which is otherwise working. I can use it in other applications (Zoom, web browsers) and it appears as "/dev/video?" (depending on if other video sources are present on the computer; I've got two to test on).

I can add the device manually, however if I do that, the UI becomes near-unresponsive, with lags of may tens of seconds. I need to reset configuration to get a usable Webcamoid again.

## Current Behavior

Webcamoid does not add the device automatically. On my laptop it does add my built-in Webcam, but not the USB one. On my Desktop, there are no other camera devices, and Webcamoid founds none.

If I try to add the device manually (configure sources -> "+" -> add in /dev/video?), the UI becomes near-unresponsive; presumably because it handles the device node as a file.

## Expected Behavior

Webcamoid should detect the device automatically. If it doesn't, there should be a way to add it manually. 

Webcamoid should realize giving /dev/video? as a source is not a regular file but a v4l device node. The UI should not hang.

## Steps to Reproduce

1.  Open Webcam with an USB webcam attached to the computer. Observe the USB webcam not being detected (problem no 1)
2.  Try to add device: Configure sources -> "+" -> type in a name and corresponding /dev/video?
3.  Observe UI becoming near-unusable as in getting stuck, with many tens of seconds of lag to do anything (problem no 2).

## Other information
This is a new installation of Webcamoid. I've tried to delete .config/Webcamoid a few times to start over. I've restarted Webcamoid with no configuration file present, with the webcam plugged in, and also plugging in adfter starting webcamoid.

lsusb -v of the webcam: https://gist.github.com/WildPenquin/bc4260b2526b62067a917c209c407226

hipersayanX, 2020-04-25:
Webcams are detected automatically, if not there is some bug related to your camera.
Try **webcamoid-git** instead, I've recently fixed some bugs related to v4l2.

cr1st1p, 2020-05-07:
I just installed webcamoid and it says no webcam detected. The webcam works fine in other programs. Using Manjaro - which is an Arch distro based.

A strace on the program shows that program actually talks with it. I'll attach parts of the strace.

[a2.log](https://github.com/webcamoid/webcamoid/files/4595018/a2.log)

cr1st1p, 2020-05-07:
Git version d3a4... seems to detect it but:

a) I can not see anything on the main screen. I have a blank canvas, like in attachment.

[image]

  Well, attachment is better than the initial window because now at least it displays in the title my webcam type and it shows 2 controls on the top left. Imagine the window *without* those

b) I figured out by moving the mouse that I have some left top menu and a right top menu. 
  It's how I saw the webcam was detected. 
  But again, I do not see anything on the main window. 

UPDATE: Hmm, usability. After I clicked on that simple single black color nothing-inspiring "icon" near the "Play sources" I also got an image :-)

hipersayanX, 2020-05-08:
@cr1st1p

>I just installed webcamoid and it says no webcam detected. The webcam works fine in other programs. 

A user already reported the same issue, the camera did not get detected because it was returning 0 fps when querying for the supported frame rates, it was already fixed.

>I can not see anything on the main screen. I have a blank canvas, like in attachment.

There is some package missing not sure which one, give me result of this:

```
LC_ALL=C pacman -Ss qt5- | grep extra/qt5- | grep installed
```

>UPDATE: Hmm, usability. After I clicked on that simple single black color nothing-inspiring "icon" near the "Play sources" I also got an image :-)

How would you improve it? I'm open to suggestions #216.

cr1st1p, 2020-05-09:
Requested output:
`
extra/qt5-3d 5.14.2-1 (qt qt5) [installed]
extra/qt5-base 5.14.2-2 (qt qt5) [installed: 5.14.2-1]
extra/qt5-connectivity 5.14.2-1 (qt qt5) [installed]
extra/qt5-declarative 5.14.2-1 (qt qt5) [installed]
extra/qt5-doc 5.14.2-1 (qt qt5) [installed]
extra/qt5-examples 5.14.2-1 (qt qt5) [installed]
extra/qt5-graphicaleffects 5.14.2-1 (qt qt5) [installed]
extra/qt5-imageformats 5.14.2-1 (qt qt5) [installed]
extra/qt5-location 5.14.2-2 (qt qt5) [installed: 5.14.2-1]
extra/qt5-multimedia 5.14.2-1 (qt qt5) [installed]
extra/qt5-networkauth 5.14.2-1 (qt qt5) [installed]
extra/qt5-quickcontrols 5.14.2-1 (qt qt5) [installed]
extra/qt5-quickcontrols2 5.14.2-1 (qt qt5) [installed]
extra/qt5-remoteobjects 5.14.2-1 (qt qt5) [installed]
extra/qt5-script 5.14.2-1 (qt qt5) [installed]
extra/qt5-sensors 5.14.2-1 (qt qt5) [installed]
extra/qt5-serialport 5.14.2-1 (qt qt5) [installed]
extra/qt5-speech 5.14.2-1 (qt qt5) [installed]
extra/qt5-svg 5.14.2-1 (qt qt5) [installed]
extra/qt5-tools 5.14.2-2 (qt qt5) [installed]
extra/qt5-translations 5.14.2-1 (qt qt5) [installed]
extra/qt5-virtualkeyboard 5.14.2-1 (qt qt5) [installed]
extra/qt5-wayland 5.14.2-1 (qt qt5) [installed]
extra/qt5-webchannel 5.14.2-1 (qt qt5) [installed]
extra/qt5-webengine 5.14.2-2 (qt qt5) [installed: 5.14.2-1]
extra/qt5-webkit 5.212.0alpha4-4 [installed: 5.212.0alpha4-3]
extra/qt5-websockets 5.14.2-1 (qt qt5) [installed]
extra/qt5-x11extras 5.14.2-1 (qt qt5) [installed]
extra/qt5-xmlpatterns 5.14.2-1 (qt qt5) [installed]
                                                   
`

Regarding the UI:
- just an observation: it looks like you implemented your own Switch control and not used Qt QML's own one?
- I think that the issue for me might come from a bad theme color combination somewhere

I also had the following problems:
- Using the akvcam - at some point my whole computer crashed and rebooted.
- I did not succeed to create a virtual cam with either v4l2loopback or akvcam - which was my main point in trying the program.

Sorry, in the end I had to switch to some other program.

But keep up the work - I liked the UI from #216 !

Regards.

hipersayanX, 2020-05-11:
@cr1st1p 

>Requested output:

That's weird, you have all packages installed, could it be a bug related to the desktop, what desktop are you using?

>just an observation: it looks like you implemented your own Switch control and not used Qt QML's own one?

Yes, I had some problems with the default themes, so I implemented the whole style following more or less [Material desing](https://material.io/components).

>Using the akvcam - at some point my whole computer crashed and rebooted.

You can report the issue [here](https://github.com/webcamoid/akvcam/issues).

**EDIT**

I have the problem of icons not showing up if I set the style provided by KDE, if I set the icon.name property it can actually find the icons, I think this is related to [this](https://github.com/webcamoid/webcamoid/issues/289#issuecomment-625436898). For some reason the controls are reading icon.name instead of icon.source.

hipersayanX, 2021-02-16:
If I remember well this was fixed in the master already.

closed 2021-02-16 by hipersayanX