webcamoid/webcamoid #290: eNixOS packaging issue

robaca, 2020-04-27:
Hi,

I just try to create a package description for webcamoid on NixOS and ran into an issue. The compilation looks fine so far (except that I did not provide all optional dependencies), but when I run the created binary, I encounter the following error on startup:

```
qt5ct: using qt5ct plugin
qt5ct: D-Bus system tray: no
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:171 Type Label unavailable
file:///nix/store/z2a28fmj0nzpaynp6z1yxgn7asvslr8v-qtquickcontrols2-5.12.7-bin/lib/qt-5.12.7/qml/QtQuick/Controls.2/Universal/qmldir:-1 plugin cannot be loaded for module ".nix.store.z2a28fmj0nzpaynp6z1yxgn7asvslr8v-qtquickcontrols2-5.12.7-bin.lib.qt-5.12.7.qml.QtQuick.Controls.Universal": Module namespace 'QtQuick.Controls.Universal' does not match import URI '.nix.store.z2a28fmj0nzpaynp6z1yxgn7asvslr8v-qtquickcontrols2-5.12.7-bin.lib.qt-5.12.7.qml.QtQuick.Controls.Universal'

qt5ct: palette support is disabled
```
The process is then hanging without showing any UI. 

As far as I can tell, the QML2_IMPORT_PATH seems to contain the correct folders for QtQuick, but I'm not an expert in QT development, so maybe s/o has an idea or can assist on this task to get it working.

My current package script looks like this:
```
{ stdenv, fetchFromGitHub, libxcb, qt5, ffmpeg-full, gstreamer, libpulseaudio, gksu, v4l-utils }:
stdenv.mkDerivation rec {
  pname = "webcamoid";
  version = "8.7.1";

  src = fetchFromGitHub {
    sha256 = "1d8g7mq0wf0ycds87xpdhr3zkljgjmb94n3ak9kkxj2fqp9242d2";
    rev = "${version}";
    repo = "webcamoid";
    owner = "webcamoid";
  };

  buildInputs = [ libxcb qt5.full ffmpeg-full gstreamer libpulseaudio gksu v4l-utils ];

  nativeBuildInputs = [ qt5.qmake qt5.wrapQtAppsHook ];

  qmakeFlags = [ 
      "Webcamoid.pro" 
      "FFMPEGINCLUDES=${ffmpeg-full}/include" 
      "FFMPEGLIBS=-L${ffmpeg-full}/lib"
      "FFMPEGLIBS+=-lavformat"
      "FFMPEGLIBS+=-lavcodec"
      "GSTREAMERINCLUDES=${gstreamer.dev}/include"
      "GSTREAMERLIBS=-L${gstreamer}/lib"
      "GSTREAMERLIBS+=-lgstaudio"
      "GSTREAMERLIBS+=-lgstvideo"
      "INSTALLQMLDIR=$(out)/lib/qt/qml"
  ];

  enableParallelBuilding = true;
...
}
```
QML2_IMPORT_PATH
```
$ echo $QML2_IMPORT_PATH                                                                                                                                   
/nix/store/qd2g08w394jzp8df75vrr618i8fgxwik-qt-full-5.12.7/lib/qt-5.12.7/qml:/nix/store/a1k6k03by47p9syrar3q44ajvklzhcn8-qtwayland-5.12.7-bin/lib/qt-5.12.7/qml:/nix/store/r6j342g6b4rl4cm9s9bjw3s8cxhd9kp3-qtquickcontrols-5.12.7/lib/qt-5.12.7/qml:/nix/store/ps629idn1w8izggri7iwbm85hk90720g-qtdeclarative-5.12.7-bin/lib/qt-5.12.7/qml
```
Generated package tree (/nix/store/a51yka6c04hn6ykjnbj2i7nz9a5mvhvj-webcamoid-8.7.1):
```
.
├── bin
│   └── webcamoid
├── lib
│   ├── avkys
│   │   ├── libACapsConvert.so
│   │   ├── libAging.so
│   │   ├── libAudioDevice.so
│   │   ├── libAudioGen.so
│   │   ├── libBlur.so
│   │   ├── libCartoon.so
│   │   ├── libChangeHSL.so
│   │   ├── libCharify.so
│   │   ├── libCinema.so
│   │   ├── libColorFilter.so
│   │   ├── libColorReplace.so
│   │   ├── libColorTap.so
│   │   ├── libColorTransform.so
│   │   ├── libConvolve.so
│   │   ├── libDelayGrab.so
│   │   ├── libDenoise.so
│   │   ├── libDesktopCapture.so
│   │   ├── libDice.so
│   │   ├── libDistort.so
│   │   ├── libDizzy.so
│   │   ├── libEdge.so
│   │   ├── libEmboss.so
│   │   ├── libEqualize.so
│   │   ├── libFaceDetect.so
│   │   ├── libFalseColor.so
│   │   ├── libFire.so
│   │   ├── libFrameOverlap.so
│   │   ├── libGrayScale.so
│   │   ├── libHalftone.so
│   │   ├── libHypnotic.so
│   │   ├── libImplode.so
│   │   ├── libInvert.so
│   │   ├── libLife.so
│   │   ├── libMatrix.so
│   │   ├── libMatrixTransform.so
│   │   ├── libMultiplex.so
│   │   ├── libMultiSink.so
│   │   ├── libMultiSrc.so
│   │   ├── libNervous.so
│   │   ├── libNormalize.so
│   │   ├── libOilPaint.so
│   │   ├── libPhotocopy.so
│   │   ├── libPixelate.so
│   │   ├── libPrimariesColors.so
│   │   ├── libQuark.so
│   │   ├── libRadioactive.so
│   │   ├── libRipple.so
│   │   ├── libScanLines.so
│   │   ├── libScroll.so
│   │   ├── libShagadelic.so
│   │   ├── libSwirl.so
│   │   ├── libTemperature.so
│   │   ├── libVideoCapture.so
│   │   ├── libVignette.so
│   │   ├── libVirtualCamera.so
│   │   ├── libWarhol.so
│   │   ├── libWarp.so
│   │   ├── libWave.so
│   │   └── submodules
│   │       ├── ACapsConvert
│   │       │   ├── libffmpegav.so
│   │       │   ├── libffmpegsw.so
│   │       │   └── libgeneric.so
│   │       ├── DesktopCapture
│   │       │   └── libqtscreen.so
│   │       ├── MultiSink
│   │       │   └── libffmpeg.so
│   │       ├── MultiSrc
│   │       │   └── libffmpeg.so
│   │       └── VideoCapture
│   │           ├── libffmpeg.so
│   │           ├── libgeneric.so
│   │           └── libv4l2sys.so
│   ├── libavkys.so -> libavkys.so.8.7.1
│   ├── libavkys.so.8 -> libavkys.so.8.7.1
│   ├── libavkys.so.8.7 -> libavkys.so.8.7.1
│   ├── libavkys.so.8.7.1
│   └── qt
│       └── qml
│           └── AkQml
│               ├── libAkQml.so
│               └── qmldir
└── share
    ├── applications
    │   └── webcamoid.desktop
    ├── icons
    │   └── hicolor
    │       ├── 128x128
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 16x16
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 22x22
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 256x256
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 32x32
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 48x48
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 64x64
    │       │   └── apps
    │       │       └── webcamoid.png
    │       ├── 8x8
    │       │   └── apps
    │       │       └── webcamoid.png
    │       └── scalable
    │           └── apps
    │               └── webcamoid.svg
    ├── licenses
    │   ├── avkys
    │   │   └── COPYING
    │   └── webcamoid
    │       └── COPYING
    └── man
        └── man1
            └── webcamoid.1.gz
```

Best regards.
Carsten

hipersayanX, 2020-04-27:
1. Just in case, these lines are not required if you use **pkg-config**:

```
      "FFMPEGINCLUDES=${ffmpeg-full}/include" 
      "FFMPEGLIBS=-L${ffmpeg-full}/lib"
      "FFMPEGLIBS+=-lavformat"
      "FFMPEGLIBS+=-lavcodec"
      "GSTREAMERINCLUDES=${gstreamer.dev}/include"
      "GSTREAMERLIBS=-L${gstreamer}/lib"
      "GSTREAMERLIBS+=-lgstaudio"
      "GSTREAMERLIBS+=-lgstvideo"
```

2. You need both packages, qtquickcontrols and qtquickcontrols2 to be installed.
3. At least in  one of the path returned by QML2_IMPORT_PATH there must be the following path: _QtQuick/Controls.2/Universal_.
4. Since you are installing the Qml files in an unconventional path, create a **qt.conf** file in the same folder as _/bin/webcamoid_ with the following content:

```
[Paths]
Imports = ../lib/qt/qml
Qml2Imports = ../lib/qt/qml
```

You can check the Qt documentation:

https://doc.qt.io/qt-5/qt-conf.html
https://doc.qt.io/qt-5/qtqml-syntax-imports.html#qml-import-path

robaca, 2020-04-29:
created a pull request on nixpkgs: https://github.com/NixOS/nixpkgs/pull/86261

closed 2020-04-29 by robaca

hipersayanX, 2020-04-30:
Nice :smile: