webcamoid/webcamoid #292: After uninstall Screen Share of Hangout crash the chrome

goyometeojorito, 2020-04-28:
After uninstall Screen Share of Hangout crash Google Chrome.
In Events Viewer of windows show, the virtual cam error (after uninstall)

## Steps to Reproduce

This video show the problem:
https://youtu.be/360mHDXGB9g

## Suggestions and tips

The uninstall don't work correctly

## Your Environment

* Webcamoid information: 8.7.1 x64
* Operating System information: Windows 10 x64

goyometeojorito, 2020-04-28:
[image]

Techassi, 2020-04-29:
You have to manually delete `C:\Program Files\AkVirtualCamera.plugin` and stop & delete the registered service `Webcamoid virtual webcam`.

goyometeojorito, 2020-04-30:
I already did that, and I even looked in the windows registry for all the
references and deleted them, but even so it has left the Google Chrome
broken and the share screen function causes it to break.
Please, the uninstall, must be clean!

Techassi, 2020-04-30:
Oh okay... I did work for me after deleting these things, but a warning keeps popping up in my software - but works.

And yes the uninstall needs to be clean!

hipersayanX, 2020-04-30:
Duplicate of #178.
[Here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-606301900) and [here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-614857556) is explained how to remove the virtual cameras, and why that happens.

@Techassi 

>And yes the uninstall needs to be clean!

I know, this issue will be fixed for sure.

closed 2020-04-30 by hipersayanX

goyometeojorito, 2020-05-01:
The bug it's only in Chrome, the other navigators works fine. But chrome
keep broken to show screen in hangouts.

hipersayanX, 2020-05-01:
@goyometeojorito is more or less the same problem, the virtual camera needs a bit of redesign there.