webcamoid/webcamoid #294: Camera and microphone destroyed

manos92, 2020-04-30:
I unistalled Webcamoid as I did not need it anymore. Now, the camera and the microphone do not work when I am trying to make a video call...

hipersayanX, 2020-04-30:
What do you mean with "destroyed"? What operating system?

manos92, 2020-05-01:
Windows 10. Trying to make a video call from messenger this appears
[cid:843566be-72dd-48c7-a08c-dc69c30186ab]
Trying to give access then this appears

[cid:2e92996a-2ee2-4cbe-8c2a-b5bc509d6d75]

The default of my camera is not this name. This name should not even exist anymore.

Furthermore, no video call even if it is in Viber, or Skype cannot use my microphone so nobody listens to me.

________________________________
Από: hipersayanX <notifications@github.com>
Στάλθηκε: Παρασκευή, 1 Μαΐου 2020 1:52 πμ
Προς: webcamoid/webcamoid <webcamoid@noreply.github.com>
Κοιν.: manos92 <em.louloudis@hotmail.com>; Author <author@noreply.github.com>
Θέμα: Re: [webcamoid/webcamoid] Camera and microphone destroyed (#294)

What do you mean with "destroyed"? What operating system?

hipersayanX, 2020-05-01:
The screenshots were not uploaded properly, can you upload again? use the ```Attach files by dragging & dropping, selecting or pasting them.``` option bellow in the comment.

>Furthermore, no video call even if it is in Viber, or Skype cannot use my microphone so nobody listens to me.

Check your microphone volume is up, beside that Webcamoid is just a capture program, it does not modify nothing related to the audio.

manos92, 2020-05-01:
[image]

[image]

This camera that appears named manolis in the Google Chrome is something that I had created using webcamoid. It does not exist anymore but cannot be changed. My microphone operates well, when it is not related with a live chat. The settings of my microphone and camera are ok in general as you can see.

[image]

[image]

[image]

hipersayanX, 2020-05-02:
@manos92 get it, you created a virtual and when removing it did not get erased properly right? it has been discussed how to fix that [here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-606301900) and [here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-614857556). Also en the **Camera and microphone allowed** you can actually select either the virtual camera or your webcam.
With the audio part sorry but don't understand greek? russian? it seems to be disabled in Chrome, it could be that is disabled because your are not using it.

manos92, 2020-05-02:
@hipersayanX  I ran the .bat file. I also have deleted everything that appears to the search of the disk C: that is related with the words akvcam, akvirtualcamera and webcamoid. I have tried to reinstall my camera. I have tried everything. Nothing works. I have tried to open a video call in Messenger from every browser and every browser shows that there is a camera named manolis and uses it as my default camera and cannot be changed.

manos92, 2020-05-02:
[image]
@hipersayanX  Maybe everything here that is related with the number 20856 has to do with the akvirtual camera? and should be deleted?

manos92, 2020-05-03:
@hipersayanX  What is your opinion?

hipersayanX, 2020-05-05:
Open **regedit** and give me screenshots of this paths:

```
Computer\HKEY_CLASSES_ROOT\CLSID\{860BB310-5D01-11d0-BD3B-00A0C911CE86}\Instance
Computer\HKEY_CLASSES_ROOT\WOW6432Node\CLSID\{860BB310-5D01-11d0-BD3B-00A0C911CE86}\Instance
```

**Don't do anything else.**

manos92, 2020-05-05:
[image]

[image]

manos92, 2020-05-05:
Thank you in advance @hipersayanX  for your time !

hipersayanX, 2020-05-05:
The **{D550C587-...}** is the virtual camera and you can delete that key, **don't delete any other key**.
That will remove the virtual camera.
For the missing webcam, the webcam is detected by the but does not appears in the **Instances**, try reinstalling the driver.

https://www.youtube.com/watch?v=RhaveEmhsbA

Is more or less the same for any webcam.

manos92, 2020-05-06:
It is ok now. Thank you very much for the detailed instructions!

https://www.youtube.com/watch?v=RhaveEmhsbA

hipersayanX, 2020-05-08:
Glad it worked :smile: 
Will try to fix it before the next release.

closed 2022-02-15 by hipersayanX