webcamoid/webcamoid #295: Virtual camera doesn't work in online camera tests or Cheese

Bernie1111, 2020-05-01:
Linux Mint 19.3 Cinnamon, v4l2loopback as a virtual camera.  But when I try to use the virtual camera in firefox on webcam test websites, it says it can't be used or authorized, even though I just clicked allow on the pop-up for it, and the websites work with my built in camera.
Tried using Cheese to see if that would recognize it and it didn't either.  Am I doing something wrong?

Bernie1111, 2020-05-02:
Even trying to use it in VLC it gives this error
Your input can't be opened:
VLC is unable to open the MRL 'v4l2:///dev/video2'

hipersayanX, 2020-05-02:
This is already documented in the wiki [wiki](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support), this is more like a [v4l2loopback](https://github.com/umlaeute/v4l2loopback/issues) problem rater than Webcamoid.

closed 2021-02-16 by hipersayanX