webcamoid/webcamoid #296: Output in v4l2 format

PonasKovas, 2020-05-02:
Hello, I want to apply some filters and effects to my camera and then output the result to my virtual v4l2loopback camera. I thought I could achieve this by "recording" in v4l2 format to the `/dev/video1` file, but I can't find the v4l2 format in the list.

hipersayanX, 2020-05-05:
Go to Preferences > Output > check **Virtual Camera** > and select the virtual camera or **Add** a new one.

PonasKovas, 2020-05-05:
Doesn't work
[image]

PonasKovas, 2020-05-05:
And if I run webcamoid as root, instead of saying `Error creating camera:` it just crashes.

hipersayanX, 2020-05-05:
>And if I run webcamoid as root

**Don't** run Webcamoid as root.

Will check.

PonasKovas, 2020-05-06:
[image]
It's also worth mentioning that the `Dummy video device (0x00000)` from the previous screenshot is a virtual camera that I want to stream to (adding a new one works too, these were just my initial plans) and it is selected as you can see, but it doesn't work - nothing gets streamed there.

kickingsawdust, 2020-05-06:
Hey, someone having the exact same problem I am :) 

I'm on opensuse 15.1 running 8.7.1 from the AppImage. If I try to add a camera it asks for system permissions, I type in the password and then the app locks up. Trying to accomplish the same thing I want to run webcamoid as a virtual camera so I can use it in zoom and OBS.

Happy to provide any details that would help. Nothing interesting is logged when this happens:
~/Downloads> ./webcamoid-8.7.1-x86_64.AppImage 
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/VirtualCamera/share/qml/main.qml:25:1: QML GridLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/VirtualCamera/share/qml/main.qml:25:1: QML GridLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/OutputConfig.qml:26:1: QML Flickable: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.

This is logged then nothing else so I go in try to save a vcam and the UI locks up that is all.
Thanks, in advance.

Bill

kickingsawdust, 2020-05-06:
Nothing interesting in /var/log/messages either:

2020-05-06T15:21:57.401449-05:00 fireball pkexec: pam_unix(polkit-1:session): session opened for user root by (uid=0)
2020-05-06T15:21:57.402279-05:00 fireball pkexec[30091]: root: Executing command [USER=root] [TTY=unknown] [CWD=/root] [COMMAND=/usr/bin/sh /tmp/Webcamoid-jPBvMl/akvcam_exec.sh]

hipersayanX, 2020-05-08:
I'm working in the virtual camera right now. I will give an update when it can be tested in the daily build, of course if I can remember this issue, else remind me later :sweat_smile:

PonasKovas, 2020-05-11:
As I understand you need to have akvcam installed to be able to create cameras right? My problem is that I can't use already created cameras with v4l2loopback

hipersayanX, 2020-05-11:
@PonasKovas 

>As I understand you need to have akvcam installed to be able to create cameras right?

You can install akvcam, or you can pass ```--vcam "/path/to/akvcam/src"``` and Webcamoid will compile and load the driver (but it won't install it).

>My problem is that I can't use already created cameras with v4l2loopback

Some programs does not detects v4l2loopback because it advertise itself as a M2M device, akvcam does not.

PonasKovas, 2020-05-12:
> Some programs does not detects v4l2loopback because it advertise itself as a M2M device, akvcam does not.

That's not what I meant. My v4l2loopback camera works perfectly, I can stream to it using `ffmpeg` and most of the apps I use recognize it correctly, the problem is on Webcamoid side.

AnimiVulpis, 2020-06-04:
@kickingsawdust 
Do you have a polkit agent running?
If not you might run into [this bug](https://gitlab.freedesktop.org/polkit/polkit/-/issues/17) which prevents creation of cameras.
[This arch wiki entry about Polkit](https://wiki.archlinux.org/index.php/Polkit#Authentication_agents) might help choose an agent.

@PonasKovas for me webcamoid could not create a second v4l2loopback virtual camera **if** there was already one defined via `modprobe`

PonasKovas, 2020-06-04:
Can't it just use an existing camera?

hipersayanX, 2020-06-07:
@PonasKovas yes, if you created the virtual camera manually Webcamoid will detect it.

PonasKovas, 2020-06-08:
It does detect it now, when I compiled from the master branch, but it still doesn't stream to it.

42demian42, 2021-04-22:
Hi.  I don't know if this is helpful, but I am having the same issue.   Arch/KDE -  "Error creating camera".  I can provide info / testing if needed.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX