webcamoid/webcamoid #297: empty file on video recording, 8.7.1 macOS

schlomo, 2020-05-03:
# README

# Reporting a problem?

## Summary

Recording a video creates an empty file

## Current Behavior

When recording a video from the webcam the output file is empty, no matter which webcam or video codec setting I use. I tried MJPEG and YUV2 in different resolutions and different codecs/formats like iPod H.264, DV etc.

The console log looks like this:

```
$ /Applications/webcamoid.app/Contents/MacOS/webcamoid
WARNING: Cannot find style "Universal" - fallback: "/Applications/webcamoid.app/Contents/Resources/qml/QtQuick/Controls/Styles/Desktop"
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/VideoCapture/share/qml/main.qml:276: TypeError: Cannot call method 'createCameraControls' of null
qrc:/VirtualCamera/share/qml/main.qml:25:1: QML GridLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Can't write header:  Error number 22 occurred
```

Taking photos and the preview both function.

With Quicktime or other tools video recording from the webcam is no problem.

## Expected Behavior

Create a video file with content. #53 seems to be similar but on Gentoo and fixed several years ago.

## Your Environment

* Webcamoid information: 8.7.1
* Operating System information: macOS 10.15.4 Catalina x86_64
* Any other useful information: exactly same behavior with the development version
* Webcam is Logitech C920

hipersayanX, 2020-05-05:
Which file format, with which video codec, with which audio codec, gives you the ```Can't write header:  Error number 22 occurred``` error message?
Give me a bounded set of options so I can test this bug quickly.

schlomo, 2020-05-05:
[image]
I didn't configure anything in detail.

Is there a way to write a detailed debug log?

hipersayanX, 2020-05-05:
Everything seems to be correct there, also thanks for testing the daily build. Will check.

>Is there a way to write a detailed debug log?

Webcamoid logging is pretty much rudimentary, what you see in the terminal is what you get, and it could be incomplete. The only way of fixing this kind of bugs is getting into the source code and tracing it until you discover where is the problem.

schlomo, 2020-05-05:
We can do a 1:1 video call if you want to look at my system

hipersayanX, 2020-05-05:
Sorry but I don't give private support.

schlomo, 2020-05-06:
Wasn't asking for support, I was offering access to the problem if you want. No worries.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX