webcamoid/webcamoid #298: Distorted Video Feed

FunkFreaksAmsterdam, 2020-05-04:
Hi, 

I make to connect my Go Pro 7 black to webcamoid, and I get the video feed in the program where I need my virtual camera, but the feed comes out like this.

[image]

Any ideas?

I'm on Mac OSX Mojave btw.

Thank you!

hipersayanX, 2020-05-05:
What is the format and resolution of the stream?

FunkFreaksAmsterdam, 2020-05-08:
I’m streaming via wifi and fffmpg, is it maybe a PAL/NTSC thing? How do I see the resolution of the stream?

hipersayanX, 2020-05-09:
>I’m streaming via wifi and fffmpg, is it maybe a PAL/NTSC thing?

I don't understand, you tested with both a Go Pro and FFmpeg?

>How do I see the resolution of the stream?

When you use FFmpeg it shows the stream information in the terminal.
For the Go Pro, try opening the stream with ffplay, it will show the same information as FFmpeg.

FunkFreaksAmsterdam, 2020-05-09:
In order to have the GoPro streaming in webcamoid, I need to run fffmpeg otherwise it won’t load.

Streaming video: h264 848x700 25fps 25tbr

is that enough informations?

hipersayanX, 2020-05-11:
@FunkFreaksAmsterdam Yes, that's enough.

>848x700

The problem is that the width of the video (848) is not multiple of 32, and Webcamoid is not handling the align properly.

FunkFreaksAmsterdam, 2020-05-11:
Yes, it was indeed that, once changed the resolution in the Camera Suite settings, now it works! Thanks for sharing the infos!!

closed 2020-05-11 by FunkFreaksAmsterdam