webcamoid/webcamoid #304: Removing generated virtual camera does nothing (Linux)

tsiemens, 2020-05-12:
## Summary
Webcamoid cannot remove akvcam devices added with webcamoid

## Current Behavior
I added a couple akvcam devices with webcamoid.

I attempted to remove one of them, but it did not appear to do anything. Neither the entry shown in webcamoid's devices dropdown (Preferences > Output) was removed, nor the relevant /dev/videox devices. No error is shown in the UI.

Restarting webcamoid did not appear to help, nor did ensuring nothing else on the system was using the devices.

I tried running webcamoid from both a gnome desktop entry, and directly from bash.
When run from bash, there was no stderr/stdout reporting some error encountered.

## Expected Behavior
In the Preferences > Output panel, clicking the "Remove" button with a created akvcam device should remove it from the system, as well as the UI.

If there was some error, I'd expect it to appear in the UI, or at least in the stderr.

## Steps to Reproduce
- Run webcamoid (in this case from the AppImage binary)
- Go to Preferences > Output
- Create an akvcam device
- (optional) test the akvcam devices works (it did)
- In Preferences > Output again, select the device from the Devices dropdown
- Click "Remove"
- Enter the root password in the prompt for pkexec

## Suggestions and tips
I have not found a solution to the issue so far. I've avoided manually removing the virtual devices, should there be some extra metadata which would be left around from webcamoid.

## Your Environment
* Webcamoid information: webcamoid-8.7.1-x86_64.AppImage
* Operating System information: Fedora 32 - 5.6.11-300.fc32.x86_64
* Any other useful information:

There were no stderr/stdout that I could see, reporting what the result from pkexec was.

Device ownership (in and out)
```
crw-rw----+ 1 root video 81, 5 May 12 10:55 /dev/video5
crw-rw----+ 1 root video 81, 6 May 12 10:55 /dev/video6
```

MuriloucoLouco, 2021-02-01:
I'm having the same problem. Anyone found temporary fix? In my case, i created a source with Media UID /media/video1, and now i can't remove or add virtual cameras. When i try to add one, it crashes, though clicking in "Remove All" delete the /etc/akvcam/config.ini folder (but nothing else happens). I've already tried to remove webcamoid and it's configs, but nothing worked. Is there a manual way to do it?

steve-vincent, 2021-05-19:
Same issue here.... 

Could not remove with Webcamoid, and even worse... after uninstalling, the devices are still there.

hipersayanX, 2021-05-19:
For those having problem removing the virtual camera, if you want to stop loading it at boot time, just delete the files:

```
rm -vf /etc/modprobe.d/akvcam.conf
```

For  unloading the module do:

```
sudo modprobe -r akvcam
```

For deleting all virtual cameras configs

```
rm -rvf /etc/akvcam
```

For uninstalling the module, depends of how did you installed it:

1. If you installed it with the installer, just run the uninstaller.
2. If you installed it manually from sources then run ```dkms remove akvcam/x.y.z --all``` (replace _x.y.z_ with akvcam version).
3. If you installed it from your distro repositories, uninstall it in the same way (ask your distro maintainers if you don't know how).

hipersayanX, 2021-05-20:
For those having this bug, could it be that you were running the virtual camera or a source while trying to remove the virtual camera?

tsiemens, 2021-05-20:
(Wups sorry for the open/close spam). I have since moved to a different linux distro and machine, and never reinstalled webcammoid, so I can't really add any more details. You'll need to collect info from steve or MuriloucoLouco.

Given that I originally stipulated that testing the camera was optional, I presumably tried both using and not using the camera.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX