webcamoid/webcamoid #306: Virtual Web Cam is not working on Windows 2012 R2 Standard

gammafour, 2020-05-16:
## Reporting a problem?

## Summary

I tried to install the 64 bit version and the 32 bit version on Windows 2012 and create a virtual cam but it does not work in Chrome, Chrome can't detect any Web Cam

## Current Behavior

I can create the virtual cams successfuly without errors, I tried with both 64 bit and 32 bit version on  Windows 2012 R2 Standard

## Expected Behavior

Virtual Web Cam should work in Chrome

## Steps to Reproduce

1- Run Webcamoid (v8.7.1)
2- Go to settings and then Output
3- Enable Virtual Web Cam and create a new one
4- Open Chrome
5- Test Web Cam using a Web Cam Test site: https://www.onlinemictest.com/webcam-test/
6- Web Cam Not working

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (v8.7.1) (64bit)
* Operating System information: (Windows Server 2012 R2)

gammafour, 2020-05-16:
Fixed:

Please follow these steps:

Windows Server 2012 R2
The WIA service is not installed during the default Windows Server 2012 installation. Before you can enable WIA, it first needs to be installed:

1. Open Server Manager and click Add roles and features below Configure this local server.
2. Move through the Wizard until you reach Features.
3. Scroll down to and expand User Interfaces and Infrastructure, then click Desktop Experience.
4. Click Add features in the dialog that displays and click Next in the Wizard.
5. Install the features and restart the server.
 

Now you can enable WIA as follows:

1. Right click Start and click Search.
2. Begin typing Administrative Tools; when this displays, click it
3. Find and double click the shortcut for Services
4. Locate and right-click "Windows Image Acquisition", then click Properties.
5. In the General tab, set the Startup type to "Automatic".
6. Click OK.
7. Right click "Windows Image Acquisition" again, then click Start.

hipersayanX, 2020-05-17:
Thanks for sharing

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX