webcamoid/webcamoid #307: webcamoid seems to not be recognized by any program on mac

jamesz1900, 2020-05-17:
good day,

When i use webcamoid it loads up fine  but the problem is i cannot use it,because in photobooth,facetime,skype and even in webbrowsers,webcamoid does not show up as a video source.
it only gives me the option to choose from my built in webcam.
i have installed another similar program called Manycam and that one seems to work fine in all the above applications.
Also when i try to create a virtual webcam it does not show up as a source to use in manycam itself.

I have disabled SIP,but it still not working

i am using macOS high sierra 10.13.6

hipersayanX, 2020-05-17:
Have tried in Mac 10.14 and is working fine, disabling SIP must be enough unless Apple introduced new walls.

jamesz1900, 2020-05-17:
Thank you for your response
i tried disabling SIP and installed different versions of webcamoid.
It must be a problem with my macbook then,any idea what it could be? since this is the only webcam software that does not seem to be found by any browser or app.

hipersayanX, 2020-05-17:
Did you tried with Chromium or Firefox?

>any idea what it could be?

The virtual plugin is not signed, that's why you must disable SIP so those program can catch the virtual camera, if you have disabled SIP and even so it does not works, it could be a bug or something else, what could it be I don't know.  
There are [some ideas](https://github.com/webcamoid/webcamoid/issues/233#issuecomment-619198980) for making a virtual camera dongle and bypass the code signing but it will take some time to implement.

Kirito-Hiro, 2020-05-24:
Hi jamesz1900, 
Maybe I am a bit too late to help out but I just read your question.
My situation, IP camera I want to use as a webcam.
Manycam is a paid solution that works great, but webcamoid is as easy as it comes and not paid.
I had the same issue as you but in my case with Zoom and I did the following.
Quit Zoom
Open Terminal and type the following line: 

codesign --remove-signature /Applications/zoom.us.app/

This was a fix for me. The odd thing is that it also helped to fix my Teams issue on virtual-cams (no idea why).
Skype for Bussiness is a no-go.

zakk4223, 2020-05-25:
FYI, It won't work even if the plugin IS signed. Library validation only allows code signed by the same team as the running app, OR "platform" binaries (i.e signed by Apple). Basically library validation completely breaks third party DAL plugins (and CFPlugins in general). I believe they 'fixed' this for Audio plugins by developing an out-of-process loading scheme, so I hope they do this with CMIO too (one day...)

Almost all Apple provided camera apps except for Quicktime Player have library validation enabled.

hipersayanX, 2020-05-25:
@zakk4223 and Microsoft is doing it even worst, the only way of creating a virtual camera in Media Foundation is writing a driver, and drivers must be signed, so...
At the end the only solution for FOSS will be relying on hardware dongles solutions :man_shrugging:

zakk4223, 2020-05-25:
Additional obnoxiousness: Zoom just released their update that re-enables virtual webcams, and they did it by pushing it to a process that doesn't have library validation, and then embedding an allow list of code signature team ids and blocking unknown ones.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX