webcamoid/webcamoid #309: Hello, i am trying to change the format resolution in Webcamoid configuration on regedit on windows

jmrsuarez, 2020-05-25:
Hello, here on México, the teachers of my country have pc with very old hardware and the internet is very slow, i am triying to use a old computer and need to create a format of very low resolution like, vga or qvga, 640 x 480 or 320 x 240, i have looked in the register  in Computer\HKEY_CURRENT_USER\Software\Webcamoid\Webcamoid, but i cannot find where i need to create a key to create or change a format for the virtual camera, i will be streaming  static images and gifs.

thanks in advance.

hipersayanX, 2020-05-25:
In the registry, the virtual camera formats has the following layout:

```
HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras\[Camera_Index]\Formats\size
HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras\[Camera_Index]\Formats\[Format_index]\format
HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras\[Camera_Index]\Formats\[Format_index]\width
HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras\[Camera_Index]\Formats\[Format_index]\height
HKLM\SOFTWARE\Webcamoid\VirtualCamera\Cameras\[Camera_Index]\Formats\[Format_index]\fps
```

_Formats\size_ must be set to the number of formats offered by the virtual camera, _index_ starts at 1 (not 0), and _format_ must be one of the [format strings supported](https://github.com/webcamoid/webcamoid/blob/90a7d3e01b3bd513aeb12d70aea101c395e5b36a/libAvKys/Plugins/VirtualCamera/src/VCamUtils/src/image/videoformat.cpp#L378) (6th column, YUY2, UYVY, RGB32 and RGB24, are the most commonly supported).

In the [daily build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily) its already possible to [customize the formats](https://github.com/webcamoid/webcamoid/issues/216#issuecomment-633117608) from the UI.

jmrsuarez, 2020-05-26:
Many thanks for the response,  i have downloaded the daily build and now i am testing it.

closed 2021-02-16 by hipersayanX