webcamoid/webcamoid #310: Fullscreen does not work with multi-monitor setup on Windows 10

theAkito, 2020-05-25:
# Reporting a problem?
Yes.

## Summary

* I have a multi-monitor setup on Windows 10.
* The app always starts between the 1st, 3rd and 5th monitor at the same time. Yes.
* Whenever I am trying to set this app to fullscreen, it teleports to the place where the app always starts, i.e. between 3 monitors, cut up through all those 3.

## Current Behavior
Fullscreen teleports the window to the limbo between 3 monitors.

## Expected Behavior

Fullscreen sets the app to fullscreen

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open app.
2.  Click _Fullscreen_ button.

## Suggestions and tips

* Detect monitors as good as the app does, when it is started. (Screen selection within app works well!)
* Fix starting location of app.

## Your Environment

* Webcamoid information:
8.7.1 x64
* Operating System information:
```
OS Name             : Microsoft Windows 10 Enterprise LTSC
OS Version          : 10.0.17763 N/A Build 17763
x64
```

hipersayanX, 2020-05-25:
Try the [daily build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily), I made some changes related to this bug some weeks ago.

theAkito, 2020-05-25:
@hipersayanX 

Way better!
However, fullscreen is still not detecting the correct dimensions of my main screen. It is quite a bit smaller than the actual screen and spans a little over the screen height, going partly invisible. Additionally, it moves partly behind the taskbar.
Resizing is wobbly, as well. (Though, before this update, it was completely unusable...)

hipersayanX, 2020-05-25:
>Resizing is wobbly, as well. (Though, before this update, it was completely unusable...)

Yeah, I've tried many possible solutions to solve that but it seems to be a Qt bug specific to Windows.

Id-ware, 2021-09-22:
yup, i can confirm this bug is still presents in the portable version 8.8.0

hipersayanX, 2021-11-11:
This is the same as #117, closing.

closed 2021-11-11 by hipersayanX