webcamoid/webcamoid #313: No way to save Video Output Options

SpamHamwich, 2020-05-31:
## Summary

The Video Output Options menu has a series of options that I am unable to save.

## Current Behavior

Changing the options in the Video Output Options menu does not persist when the menu is closed an there is no button to save changes.

## Expected Behavior

The changed settings should either update as they are changed or have a save button to commit the changes. 

## Steps to Reproduce

1.  Open Video Menu and navigate to the "Outputs" tab
2.  Click "Add Output" and create new output
3.  Click on newly created Virtual Camera output
4.  Click the switch to toggle "Horizontal mirror", change Aspect ratio to "keep", etc.
5.  No save button exists, ctrl-s does nothing, hitting enter does nothing, clicking the x to close the window discards changes. 

[image]

## Suggestions and tips

As far as I can tell, this is the only tab with both modal dialogues that open (i.e. the Edit button) and settings that appear to be visually designed to change immediately like the input video source options. I personally think the video output options should update the relevant settings immediately to mimic the input options behavior. The Edit button is a little bit confusing in this context because it makes me question whether I'm actually editing the video output at all when i update the scaling, aspect ratio, etc. It appears to be a leftover idea from an earlier UI redesign and might function better as an "Edit Formats" button with the name update extracted into it's own dialogue. 

## Your Environment

* Webcamoid information: Version 8.7.1 (Using Qt 5.14.2) 
* Operating System information: Manjaro (5.4.40-1) (x86_64 GNU/Linux)
* Any other useful information: Installed through pacman, webcamoid-git, since the standard webcamoid in the repo seemed to be an older version UI and wasn't functioning correctly with the virtual camera outputs.

hipersayanX, 2020-06-01:
>The Video Output Options menu has a series of options that I am unable to save.

I've not wrote the code to save those options and have not tested yet.

>5. No save button exists, ctrl-s does nothing, hitting enter does nothing, and clicking the x to close the settings window or closing the application discards the changes.

The changes are applied immediately, no confirmation is required.

>and might function better as an "Edit Formats" button **with the name update extracted into it's own dialogue**.

I don't understand this part, can you explain further please?

svanschalkwyk, 2022-04-06:
Same here. Great app. 
If one could save.

ilya-levin-lokalise, 2022-04-11:
@hipersayanX hello. I've noticed that video output settings are not persisted after system reboot (Ubuntu). Is it intended and are there any plans to fix it?

hipersayanX, 2022-04-12:
@ilya-levin-lokalise 

>Is it intended and are there any plans to fix it?

its not implemented, it will be implemented soon or later when I've some time or someone send a PR.