webcamoid/webcamoid #316: FEATURE REQEUST: Face Tracking

cjbarth, 2020-06-06:
Your face detection plugin works well, but it would be nice if that could automatically pan and crop the video feed as well, similar to what Nvidia recently released for OBS: https://appuals.com/nvidia-ar-sdk-debuts-in-popular-obs-plugin-streamfx-allowing-face-detection-through-standard-webcam/.

Ideally, this would be something that let you specify a padding around the faces to keep in frame. It seems that since you already have the face detection plug-in, this should be relatively easy. I'm a developer, so if there is interest, I'd be happy to take a look at doing this, but a quick point in the right direction would save me a lot of effort.

hipersayanX, 2020-06-07:
>I'm a developer, so if there is interest, I'd be happy to take a look at doing this, but a quick point in the right direction would save me a lot of effort.

You can add a function

```
Q_INVOKABLE QVector<QRect> detect(const AkVideoPacket &packet) const;
```

to FaceDetectElement and reuse [this code](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/libAvKys/Plugins/FaceDetect/src/facedetectelement.cpp#L192), then you create a new plugin and reuse FaceDetect, [here](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/libAvKys/Plugins/Fire/src/fireelement.cpp#L81) is an example on reusing plugins.

cjbarth, 2020-06-08:
Excellent! I've cloned the code and am trying to set up the build environment and I've followed the directions as best I could and ran into a few errors trying to get it to build for the first time. I'm quite sure that the problem is that I'm missing some Qt component. The problem is that the Qt components that are listed as required in the Wiki aren't listed anywhere in the installer.

I used the Qt dependency directions found at https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install.
That lead me to download Qt Online Installer from https://www.qt.io/download-open-source
I installed the following:

- Qt 5.12.2
  - MSVC2017 64-bit
  - Sources
  - Qt Debug Information Files
 - Developer and Designer Tools
  - Qt Creator 4.12.2. CDB Debugger Support
  - Debugging Tools for Windows

Here is the output from trying to run the project from Qt Creator:
```txt
16:39:23: Starting C:\Code\GitHub\build-Webcamoid-Desktop_Qt_5_12_2_MSVC2017_64bit-Debug\StandAlone\debug\Qt5.12.2\cl\x86_64\bin\webcamoid.exe ...
QML debugging is enabled. Only use this in a safe environment.
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:430 Type VideoEffectsPanel unavailable
qrc:/Webcamoid/share/qml/VideoEffectsPanel.qml:48 Type VideoEffectsList unavailable
qrc:/Webcamoid/share/qml/VideoEffectsList.qml:24 Type ScrollView unavailable
qrc:/Webcamoid/share/themes/WebcamoidTheme/ScrollView.qml:38 Non-existent attached object

qt.network.ssl: QSslSocket::connectToHostEncrypted: TLS initialization failed
Error requesting latest version: No response
```

Any tips for getting this set up?

hipersayanX, 2020-06-08:
The error complaining of this [this](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/StandAlone/share/themes/WebcamoidTheme/ScrollView.qml#L38) and [this](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/StandAlone/share/themes/WebcamoidTheme/ScrollView.qml#L46) lines, even though the documentation says it's [perfectly fine and supported](https://doc.qt.io/qt-5.12/qml-qtquick-controls2-scrollbar.html#horizontal-attached-prop), is the same as #234. Changing [version numbers](https://github.com/webcamoid/webcamoid/issues/234#issuecomment-605378882) won't help. So I've not any answer on how to solve that (Qt bug?).
Try installing Qt 5.14 or 5.15 instead, or much better just install [MSYS2](https://www.msys2.org/) and install Qt and FFmpeg [from there](https://github.com/webcamoid/webcamoid/blob/08f786d6860f137cecd654da245720bc7688cc3f/ports/ci/appveyor/install_deps.sh#L21).

cjbarth, 2020-06-13:
Upgrading to Qt 4.15 and MSVC 2019 resolved the issues, though I still get 18 warnings on compile. I don't seem to see any of the plugins or even get my webcam detected, but I'll try to work through those issues. I'm probably missing some step in the getting started.

hipersayanX, 2020-06-14:
Depends on what those warnings says, if it throws deprecated warning is ok, those must be fixed later.

cjbarth, 2020-06-16:
I've gotten most of this working, but there is an important piece that still isn't working. I tried several things, all with no success.

1. I tried adding the `detect` function like you proposed, but there is no way to call it from another plugin because I have to pass something to the function. I tried using a property to pass the frame and then remove the arguments from the `detect` function, but then I couldn't decode/cast the return type of `QVector<QRect>` from `QVariant`
2. I then tried using an event emitter to just emit when I found a face by calling `iStream` like I saw in the other examples. Sure enough, that caused a detect to work just fine, but I could get the `QObject::connect` to connected up the `QVector<QRect>`. I was able to `connect` a different emitted event that just emitted the count of faces as an `int`, but not the more complex type.

Option 2 is current pushed up to my fork. If you run the code you'll see a number in the middle of the image of the count of the faces, but I can't figure out how to get the `QVector<QRect>` back, even if I use the exact same code, which I've left committed to my branch.

Any thoughts? The only solution I can see right now would be to unpack the `QVector<QRect>` to some primitives and then emit those as events and work with that, but that seems unnecessary. Then again, I don't know the message passing ability of Qt.

cjbarth, 2020-06-28:
I've posted a PR for this change. I'm interested in feedback. I've left some debugging code in to show a box around the detected face+margin+padding for testing and validation purposes. Of course, this will be removed entirely before the code is merged.

Swicegood, 2020-07-20:
@cjbarth I also saw the need for this and was inspired by what the Facebook Portal does. I'm not an experienced developer but I decided to take a crack at it, and made a program that grabs a webcam follows the face and feeds into OBS using the Gstreamer plugin. [https://github.com/Swicegood/camshift](url). 
I'd would be more than happy to use your solution, though. 

For clarity, however. Is your solution finished?

I also was not aware that of the nvidia plugin for OBS.

cjbarth, 2020-07-20:
My code is finished and is awaiting PR review.

I will have a look at your code though as there might be some opportunities for idea exchange.

samliddicott, 2020-07-23:
are these downloadable as plugins to try in a daily build?

Swicegood, 2020-07-23:
I tried on Windows, Mac and Linux

The most recent daily build for Linux works out of the box, complete with
FaceTracking.

Mac daily build on Catalina 10.15.3 runs but no window shows up at all.

There has not been a daily build for Windows since the code was merged. But
I have successfully built it myself for Win 10 64 using the wiki. Face
Tracking plugin seems to work pretty well although I cannot play videos and
under preferences video playback and recording  there is no
driver available. I'm assuming it should be  ffmpeg. I have included the
code line giving in the guide specifying the paths of the FFMPEG libs, etc.
FaceTrasking + Virtual camera works in OB Studio.
Also I could not "deploy" or make the installer. Running deploy.py,  I'm
getting the error message:  /usr/bin/bash: -c: line 1: syntax error:
unexpected end of file
mingw32-make: *** [Makefile:84: sub-libAvKys-install_subtargets-ordered]
Error 1

Perhaps that is why there was no automated daily build.

Hopefully I'm just doing something wrong. If anybody knows please let me
know.

Swicegood, 2020-07-25:
I managed to make a Windows portable build by picking and choosing files manually, Editing webcamoid.bat and webcamoid.ini.
I can share it with you if your'e interested.
It works with my Logitech but not with my built-in cameras.

hipersayanX, 2020-07-26:
@samliddicott 

>are these downloadable as plugins to try in a daily build?

The plugin is included in the daily build.

@Swicegood 

>I cannot play videos and under preferences video playback and recording  there is no driver available. I'm assuming it should be  ffmpeg. I have included the code line giving in the guide specifying the paths of the FFMPEG libs, etc. FaceTrasking + Virtual camera works in OB Studio.

The wiki is a bit outdated, right now the easier way to setup the development environment for Webcamoid in Windows is using [MSYS2](https://github.com/webcamoid/webcamoid/blob/8950d6c034f38539882f271097912572636453b7/ports/ci/appveyor/install_deps.sh#L22). You don't need to specify any library in Qmake, pkg-config do all the work.

> Also I could not "deploy" or make the installer. Running deploy.py,  I'm getting the error message:  /usr/bin/bash: -c: line 1: syntax error:
> unexpected end of file mingw32-make: *** [Makefile:84: [...]

Will check later.

>I managed to make a Windows portable build by picking and choosing files manually, Editing webcamoid.bat and webcamoid.ini.
I can share it with you if your'e interested.

Only share source code, not binaries. Downloading binaries from untrusted sources is a bad idea.

cjbarth, 2020-08-16:
I would love to test this feature a little better in a packaged form, but I find that the nightly builds are still not working. Is there something I can do to help with that?

hipersayanX, 2020-08-16:
@cjbarth I'm working in Mac right now, i haven't tested in Windows yet, but Webcamoid may work fine running from Qt Creator, I won't work in the packaging issue until I finish with the virtual camera driver (and that will take some ~~a lot~~ time).

>Is there something I can do to help with that?

Is there an error message when building Webcamoid o when executing it?

hipersayanX, 2020-09-23:
@Swicegood 

>Mac daily build on Catalina 10.15.3 runs but no window shows up at all.
>Which is the same as issue #327

The problem may be fixed now, [the build](https://travis-ci.org/github/webcamoid/webcamoid/jobs/729672787) may be ready in a few hours.

@cjbarth I've fixed all problems with daily builds, it may work fine now.

Swicegood, 2020-09-23:
I don't know what is wrong with my installation. I'm still getting:

QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not
installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not
installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not
installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not
installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not
installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not
installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not
installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not
installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not
installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not
installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not
installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not
installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not
installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not
installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not
installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not
installed

hipersayanX, 2020-09-24:
@Swicegood checked it again, this time it may work fine :sweat_smile: , the search paths in the qt.conf file were wrong.

>I have almost got it working with building it  on MacOS. Just the effects list is empty and the camera does not show up. I suspect that I have left out some dependency.

You must pass ```-r -p webcamoid-build-directory/libAvKys/Plugins``` to the command line.

Swicegood, 2020-09-24:
Yes it works now, both with self-building and pre-built. Thanks!

With the caveat that adding a virtual camera does nothing. I will start another issue for that.

alex-caelus, 2021-04-11:
Hi! I see that the feature has been implemented and PR #329 merged to master. But I'm running version 8.8.0 and I don't see it in the plugin list. I do see face detection plugin though.

hipersayanX, 2021-04-11:
@alex-caelus is implemented in the daily build and will be released with Webcamoid 9.

alex-caelus, 2021-04-11:
Ah, I see @hipersayanX. I was confused by the fact that I saw the merge to master was a while ago. I didn't know about your release process. Thnx

closed 2022-02-15 by hipersayanX