webcamoid/webcamoid #322: Unable to start video stream of virtual camera

akadev2, 2020-06-13:
I am utilizing webcamoid's virtual camera for my app on Mac.
The app captures webcam video, process that video and create virtual camera.
I was able to instantiate it and it shows up as an available device on zoom and skype.
The problem is it only shows a placeholder image of webcamoid.
I tried to send frames to it but I cannot stream the video.
I am not sure which method I should call to start the stream.
Any help would be appreciated.

hipersayanX, 2020-06-14:
Post your code.

akadev2, 2020-06-18:
AkPacket packet({"video/x-raw"});
packet.setCaps(AkVideoCaps(AkVideoCaps::Format_nv12, input->width(), input->height(), AkFrac(24, 1), 1));
input->map(QAbstractVideoBuffer::ReadOnly);

packet.setBuffer( QByteArray((const char*)input->bits(), input->mappedBytes()) );
input->unmap();

emit oStream(packet);

This is the code I tried last time but it didn't work

hipersayanX, 2020-06-18:
If you want to use the plugin, the code is more or less like this.

```
auto virtualCamera = AkElement::create("VirtualCamera")};

// Select an output device
auto medias = cameraOutput->property("medias").toStringList();

// and set it.
virtualCamera->setProperty("media", medias.first());

/* Set the output stream as RGB24 640x480 24 FPS.
 * Only RGB24 and BGR24 formats are supported, the driver will convert the 
 * frames to the required format in the other side.
 */
int fps = 24;
AkVideoCaps caps(AkVideoCaps::Format_rgb24, 640, 480, {fps, 1});
QMetaObject::invokeMethod(virtualCamera.data(), "clearStreams");
QMetaObject::invokeMethod(virtualCamera.data(), "addStream",
                          Q_ARG(int, 0),
                          Q_ARG(AkCaps, AkCaps(caps)));

// Set a stream ID for the packets.
auto streamId = Ak::id();

// Start streaming.
virtualCamera->setState(AkElement::ElementStatePlaying);

// Send 1000 frames.
for (int i = 0; i < 1000; i++) {
    /* Passing caps to the constructor, AkVideoPacket will reserve a buffer 
     * with enough size for the frame data.
     * The other option is leaving the contructor blank and set packet.buffer() 
     * if the buffer is pre-allocated.
     */
    AkVideoPacket packet(caps);

    // Copy data to the buffer
    memcpy(packet.buffer().data(), rawFrameData, packet.buffer().size());
    
    packet.id() = streamId;
    
    /* 'pts' must increase with the time.
     * Frame time in seconds = pts * timeBase.
     */
    packet.pts() = i;
    packet.timeBase() = {1, fps};
    packet.index() = 0;
    
    // Send the packet.
    cameraOutput->iStream(packet);

    // Simulate a delay. Not required in a real environment.
    QSleep::msleep(1000 / fps);
}

// Stop streaming.
virtualCamera->setState(AkElement::ElementStateNull);
```

You can check [videolayer](https://github.com/webcamoid/webcamoid/blob/3dd917b1938f73fc1ccba6ca8dc762433f860ef6/StandAlone/src/videolayer.cpp#L67) source code if you need an example.

akadev2, 2020-06-21:
Alright. I will check it out

akadev2, 2020-06-28:
I am still unable to see the stream.
In contrary, my mac does not detect webcam anymore. I am not sure what I did wrong for that.
Here's my code modified according to your suggestion.

m_virtualCamera = AkElement::create("VirtualCamera");
m_virtualCamera->setState(AkElement::ElementStateNull);

const QDir applicationDir(QCoreApplication::applicationDirPath());
QString driverPath = QDir::cleanPath(applicationDir.absoluteFilePath("../Resources")).replace('/', QDir::separator());
//m_virtualCamera->addDriverPath(driverPath);

QMetaObject::invokeMethod(m_virtualCamera.data(), "addDriverPath", Q_ARG(QString, driverPath));
QMetaObject::invokeMethod(m_virtualCamera.data(), "createWebcam", Q_ARG(QString, "Magic Cam"));

auto medias = m_virtualCamera->property("medias").toStringList();
m_virtualCamera->setProperty("media", medias.first());
//m_virtualCamera->setProperty("media", "Magic Cam");
//AkElement::link(m_filter, m_virtualCamera.data());

/* Set the output stream as RGB24 640x480 24fps.
 * Only RGB24 and BGR24 formats are supported, the driver will convert the frames to the
 * required format in the other side.
 */

int fps = 24;
caps = AkVideoCaps(AkVideoCaps::Format_rgb24, 1280, 720, AkFrac(fps, 1), 1);

QMetaObject::invokeMethod(m_virtualCamera.data(),
                          "clearStreams");
QMetaObject::invokeMethod(m_virtualCamera.data(),
                          "addStream",
                          Q_ARG(int, 0),
                          Q_ARG(AkCaps, caps));
streamId = Ak::id();

if (m_virtualCamera->setState(AkElement::ElementStatePlaying))
	qDebug() << "setState OK";
else
{
	qDebug() << "setState falhou";
}

... ... ...
... ... ...

QVideoFrame VirtualCameraFilterRunnable::run(QVideoFrame *input, const QVideoSurfaceFormat &surfaceFormat, RunFlags flags) {
    
	//m_filter->forwardFrame( input );
    cv::Mat matBGR(input->height(), input->width(), CV_8UC3);

	input->map(QAbstractVideoBuffer::ReadOnly);

	cv::Mat frameImage(input->height() + input->height()/2, input->width(), CV_8UC1, input->bits());
	cv::cvtColor(frameImage, matBGR, cv::COLOR_YUV2BGR_NV12);

	AkVideoPacket packet(caps);

	packet.id() = streamId;
	packet.pts() = frameCounter++;
	packet.timeBase() = {1, 24};
	packet.index() = 0;

	memcpy( packet.buffer().data(), (const char*)matBGR.data, matBGR.total() * matBGR.elemSize() );
	input->unmap();
	(*m_virtualCamera)(std::move(packet));

    return *input;
}

hipersayanX, 2020-06-30:
>I am still unable to see the stream.
>In contrary, my mac does not detect webcam anymore. I am not sure what I did wrong for that.

Citing from Apple documentation:

>**Notarize Plug-ins**
>
>In macOS 10.15 and later, apps can load quarantined plug-ins — those downloaded from the internet or transferred with AirDrop — only if the plug-in is notarized. For plug-ins that aren’t notarized, the user must explicitly approve the plug-in by opening the System Preferences app and navigating to the General tab in Security & Privacy.

You can read more [here](https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution) (I can't help you with this since I haven't done it before).

Alternatively you can disable [SIP](https://support.studionetworksolutions.com/hc/en-us/articles/115003839246-How-to-disable-Systems-Integrity-Protection-SIP-in-macOS) and [disable library validation and remove entitlements](https://github.com/andrewodri/iphone-to-macos-camera/issues/7).

>Here's my code modified according to your suggestion.
>It would be great if you can spot something wrong or missing.

I can't see nothing wrong with your code, if you have done everything right it must be working. If the frames are not being send you can trace them: [1](https://github.com/webcamoid/webcamoid/blob/b8efd2b2636f5b3396c2756cdfa6177ec9320911/libAvKys/Plugins/VirtualCamera/src/cmio/VCamIPC/src/ipcbridge.mm#L1010), [2](https://github.com/webcamoid/webcamoid/blob/b8efd2b2636f5b3396c2756cdfa6177ec9320911/libAvKys/Plugins/VirtualCamera/src/cmio/Assistant/src/assistant.cpp#L1194), [3](https://github.com/webcamoid/webcamoid/blob/b8efd2b2636f5b3396c2756cdfa6177ec9320911/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera/src/plugininterface.cpp#L321), [4](https://github.com/webcamoid/webcamoid/blob/b8efd2b2636f5b3396c2756cdfa6177ec9320911/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera/src/stream.cpp#L268).

hipersayanX, 2021-02-16:
Check [akvirtualcamera](https://github.com/webcamoid/akvirtualcamera/issues/3) instead of the old driver.

closed 2021-02-16 by hipersayanX