webcamoid/webcamoid #323: There is not sound on video in virtual webcam

epiixgd76, 2020-06-15:
When i put a video on the virtual webcam there is not sound  

(i use akvcam)

hipersayanX, 2020-06-16:
There is no sound because is a virtual camera, not a virtual microphone.

jhnlmn, 2020-06-28:
I was also confused by this question. Most Webcams have audio and video. Since Webcamoid is a full featured webcam suite, then, I thought, it must have audio as well. And, indeed, there is Audio tab in Webcamoid gui, which is even more confusing.
Could you, please, clarify the extent of audio support in Webcamoid?
I really need to open a virtual Webcam in the browser and both video and audio are required. Is it possible?

hipersayanX, 2020-06-28:
>Most Webcams have audio and video. Since Webcamoid is a full featured webcam suite, then, I thought, it must have audio as well.

Physical webcams that come with a mic has both, a sound card and a camera integrated, each one works with different APIs, for example, in GNU/Linux the API for the camera is V4L2, while the API for the sound card is ALSA, if you want to implement a fake webcam with a mic, you must implement both, the V4L2 and the ALSA APIs, Webcamoid only implements the first, the camera API. Also, not all webcams comes with an integrated mic.
And if you were asking, yes, Mac and Windows does the same thing with their own APIs.
Also, it make no sense to implement the virtual mic since audio effects are not implemented yet, these will be implemented in the future but that will take a very long long looooooooooooong time, since there are other priorities first.

>And, indeed, there is Audio tab in Webcamoid gui, which is even more confusing.
>Could you, please, clarify the extent of audio support in Webcamoid?

Webcamoid can capture audio from mic and send audio to the speakers when playing a video.

>I really need to open a virtual Webcam in the browser and both video and audio are required. Is it possible?

Possible yes, but it must be implemented, and as said before, it will take a quite long time.

closed 2021-02-16 by hipersayanX