webcamoid/webcamoid #325: qml errors in terminal Xubuntu 18.04LTS

galenthurber, 2020-06-17:
# Reporting a problem?

## Summary

errors showing in terminal when running daily build 20200617

## Current Behavior

while starting program the following errors show in terminal
qrc:/Webcamoid/share/qml/main.qml:575:5: QML Settings: Failed to initialize QSettings instance. Status code is: 1
qrc:/Webcamoid/share/qml/UpdatesDialog.qml:77:5: QML Settings: Failed to initialize QSettings instance. Status code is: 1
qrc:/Webcamoid/share/qml/UpdatesConfig.qml:187:5: QML Settings: Failed to initialize QSettings instance. Status code is: 1
qrc:/Webcamoid/share/qml/UpdatesConfig.qml:181: TypeError: Property 'projectIssuesUrl' of object MediaTools(0x7ffcafd782c0) is not a function

## Expected Behavior

no errors

## Steps to Reproduce

start program

## Your Environment

20200617
Xubuntu 18.04LTS 64
4.15.0-106-generic #107-Ubuntu SMP Thu Jun 4 11:27:52 UTC 2020 x86_64 x86_64 x86_64 

Webcams lsusb
Bus 002 Device 012: ID 534d:0021  (AV to USB Easier Capture Macrosil, not detected)
Bus 001 Device 006: ID 046d:0836 Logitech, Inc. B525 HD Webcam
Bus 004 Device 004: ID 046d:08da Logitech, Inc. QuickCam Messanger (not detected often)

Also.
B525 shows up twice in video source list

hipersayanX, 2020-06-17:
> while starting program the following errors show in terminal
> qrc:/Webcamoid/share/qml/main.qml:575:5: QML Settings: Failed to initialize QSettings instance. Status code is: 1 [...]

Fixed in the latest commit.

>Webcams lsusb
>Bus 002 Device 012: ID 534d:0021 (AV to USB Easier Capture Macrosil, not detected)
>Bus 001 Device 006: ID 046d:0836 Logitech, Inc. B525 HD Webcam
>Bus 004 Device 004: ID 046d:08da Logitech, Inc. QuickCam Messanger (not detected often)

Run the following commands in those devices and give me the output:

```
v4l2-ctl -d /dev/videoX --all -L
v4l2-compliance -d /dev/videoX -f
```

galenthurber, 2020-06-19:
sudo apt install v4l-utils

[1notes.txt](https://github.com/webcamoid/webcamoid/files/4805564/1notes.txt)

hipersayanX, 2020-06-22:
@galenthurber thanks, will check.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX