webcamoid/webcamoid #326: Virtual Camera working, but not showing output in Windows 10

tonyridal, 2020-06-22:
[image]

This is Windows 10 - I've seen a similar problem reported on Macs, but no applicable solution.

I'm 95% of the way to getting this to do what I need - the attached screenshot should explain all.

I've setup the matrix transform to flip the image as desired - this is working fine in the Webcamoid app. I've setup a virtual camera, called "flipped webcam 2", which Microsoft Teams can see fine and use.

What's not working is Webcamoid is not outputting to the virtual camera - I'm getting the "Enable virtual camera" message, even though it already is enabled, even when I look at it in the Webcamoid app (second screenshot)… 

Any help gratefully received!
[image]

An-land, 2020-07-16:
I've got the exact same issue, both on kde neon and on windows 10. I've tried with both of the available drivers, but couldn't figure out how to get it to work. I tried to find a guide on how to make the virtual camera use the webcam as its input, but I was unable to find one.

anchithAcharya, 2020-09-18:
@tonyridal Try pressing the stop button at the bottom left, and then 'play' it again. That worked for me.

sijanec, 2020-12-14:
Yup, can confirm @anchithAcharya 's solution on Windows 10.0.18363 Pro. The problem happened after the latest Windows upgrade.

Shamrockie, 2020-12-29:
I'm having the same issue as well, however stopping the camera and turning it back on did not work for me.

rajesh-smartwebtech, 2021-05-01:
yes not working in win 10 same issue here

Fusseldieb, 2021-10-04:
This software is beyond broken on W10 Pro 19043

[image]
Virtual Webcam remains empty

When selecting any mp4 file as a source, when pressing play simply crashes the program and hinders it from running ever again until I change the filename of that file

No Webcam appears in VLC

And it takes ages to start up

A bummer...

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX