webcamoid/webcamoid #327: Missing QT Components, Mac 10.14.6

trevortos, 2020-06-23:
Ive been trying the dailies for a few weeks now, and it seems that there is a few missing QML modules.

QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "Qt.labs.settings" is not installed
qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22:1: module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed

Ive tried using macdeployqt to replace the libs within the package, and that doesnt seem todo anything.
I also tried building it myself on 10.14 Mojave with the newest QT stuff from Brew, and I was having the same types of issues, never could get it to launch at all.

hipersayanX, 2020-06-23:
Are you compiling 8.7.1? because master branch is working fine, I'm working on it now.

trevortos, 2020-06-23:
Yes I was trying to build 8.7.1 on OSX 10.14, without much luck. The instructions need to be updated for the newer QT Creator.

But those errors are binaries built from https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files/mac

hipersayanX, 2020-06-27:
Can confirm, it seems to be a problem with deployment since on development is working fine.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX