webcamoid/webcamoid #330: Privilege escalation

StefanBruens, 2020-07-01:
https://github.com/webcamoid/webcamoid/blob/90a7d3e01b3bd513aeb12d70aea101c395e5b36a/libAvKys/Plugins/VirtualCamera/src/v4l2sys/VCamIPC/src/ipcbridge.cpp#L861

It is possible to replace the script by a arbitrary malicious script here, which is executed with root permissions.

hipersayanX, 2020-07-02:
I need to execute all those commands at once as root, what solution do you suggest?

samliddicott, 2020-07-23:
It's also possible to replace bashrc with one that sets an alias to sudo or ssh. I'm not sure what the point of this report is.

Alternatives: Build the scripts into the program as string constants so it's hard to edit them.

hipersayanX, 2020-07-26:
@samliddicott 

>Alternatives: Build the scripts into the program as string constants so it's hard to edit them.

I've thought in that option and it does not looks like a good idea feeding a bunch of commands to the sudo utility, escaping characters and so on.
The solution that I've thought is adding  a manager to akvcam, then when I want to reconfigure the cameras Webcamoid will call the manager like:

```
sudo manager load config.ini
```

The manager will then copy config.ini, and reload the driver. The same solution applies for Windows and Mac.

hipersayanX, 2022-01-08:
At the end, the solution was much "simpler" than I thought, and works for both drivers, I'll release this and all patches together with Webcamoid 9.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX