webcamoid/webcamoid #331: Crash with screen + virtual camera on Linux

City-busz, 2020-07-07:
When I try to use webcamoid to feed the screen to a v4l2loopback virtual camera on Arch Linux, I get a crash as soon as I click on the "Virtual camera" checkbox. The virtual camera is already created, and it works if I feed from file.

Steps to reproduce:
1. Select "Screen 0" as source.
2. Enable the "Virtual camera" checkbox.

```
       Message: Process 29474 (webcamoid) of user 1000 dumped core.
                
                Stack trace of thread 29474:
                #0  0x00007faf467dc355 raise (libc.so.6 + 0x3c355)
                #1  0x00007faf467c5853 abort (libc.so.6 + 0x25853)
                #2  0x00007faf4681f878 __libc_message (libc.so.6 + 0x7f878)
                #3  0x00007faf46826d3a malloc_printerr (libc.so.6 + 0x86d3a)
                #4  0x00007faf46829e1c _int_malloc (libc.so.6 + 0x89e1c)
                #5  0x00007faf4682b5a9 malloc (libc.so.6 + 0x8b5a9)
                #6  0x00007faf4781c451 _ZN10QImageData6createERK5QSizeN6QImage6FormatE (libQt5Gui.so.5 + 0x18a451)
                #7  0x00007faf4781c66d _ZN6QImageC2ERK5QSizeNS_6FormatE (libQt5Gui.so.5 + 0x18a66d)
                #8  0x00007faf4781c6b4 _ZN6QImageC1EiiNS_6FormatE (libQt5Gui.so.5 + 0x18a6b4)
                #9  0x00007faf4781cf64 _ZNK6QImage4copyERK5QRect (libQt5Gui.so.5 + 0x18af64)
                #10 0x00007faf42ef6dd5 n/a (libQt5XcbQpa.so.5 + 0x5add5)
                #11 0x00007faf42ee4604 _ZNK10QXcbScreen10grabWindowEyiiii (libQt5XcbQpa.so.5 + 0x48604)
                #12 0x00007faf477f95c1 _ZN7QScreen10grabWindowEyiiii (libQt5Gui.so.5 + 0x1675c1)
                #13 0x00007faf2987459f _ZN11QtScreenDev9readFrameEv (libqtscreen.so + 0xa59f)
                #14 0x00007faf46e48a26 n/a (libQt5Core.so.5 + 0x2eaa26)
                #15 0x00007faf46e4ca9b _ZN6QTimer7timeoutENS_14QPrivateSignalE (libQt5Core.so.5 + 0x2eea9b)
                #16 0x00007faf46e3e10f _ZN7QObject5eventEP6QEvent (libQt5Core.so.5 + 0x2e010f)
                #17 0x00007faf47ec1702 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent (libQt5Widgets.so.5 + 0x15c702)
                #18 0x00007faf46e117ba _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent (libQt5Core.so.5 + 0x2b37ba)
                #19 0x00007faf46e696b5 _ZN14QTimerInfoList14activateTimersEv (libQt5Core.so.5 + 0x30b6b5)
                #20 0x00007faf46e69f9a n/a (libQt5Core.so.5 + 0x30bf9a)
                #21 0x00007faf4584b43c g_main_context_dispatch (libglib-2.0.so.0 + 0x5243c)
                #22 0x00007faf458991d9 n/a (libglib-2.0.so.0 + 0xa01d9)
                #23 0x00007faf4584a221 g_main_context_iteration (libglib-2.0.so.0 + 0x51221)
                #24 0x00007faf46e6a331 _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE (libQt5Core.so.5 + 0x30c331)
                #25 0x00007faf46e1013c _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.so.5 + 0x2b213c)
                #26 0x00007faf46e185c4 _ZN16QCoreApplication4execEv (libQt5Core.so.5 + 0x2ba5c4)
                #27 0x000055ad04448f64 n/a (webcamoid + 0x10f64)
                #28 0x00007faf467c7002 __libc_start_main (libc.so.6 + 0x27002)
                #29 0x000055ad0444917e n/a (webcamoid + 0x1117e)
                
                Stack trace of thread 29483:
                #0  0x00007faf1c30f246 n/a (n/a + 0x0)
                #1  0x00007faf1d3c5026 n/a (swrast_dri.so + 0x637026)
                #2  0x00007faf1d3e5d16 n/a (swrast_dri.so + 0x657d16)
                #3  0x00007faf1d3e5e9e n/a (swrast_dri.so + 0x657e9e)
                #4  0x00007faf1d3c55b8 n/a (swrast_dri.so + 0x6375b8)
                #5  0x00007faf465bc422 start_thread (libpthread.so.0 + 0x9422)
                #6  0x00007faf4689fbf3 __clone (libc.so.6 + 0xffbf3)
                
                Stack trace of thread 29475:
                #0  0x00007faf4689505f __poll (libc.so.6 + 0xf505f)
                #1  0x00007faf4362c63b n/a (libxcb.so.1 + 0xc63b)
                #2  0x00007faf4362e37b xcb_wait_for_event (libxcb.so.1 + 0xe37b)
                #3  0x00007faf42ef9c51 n/a (libQt5XcbQpa.so.5 + 0x5dc51)
                #4  0x00007faf46c2ce0f n/a (libQt5Core.so.5 + 0xcee0f)
                #5  0x00007faf465bc422 start_thread (libpthread.so.0 + 0x9422)
                #6  0x00007faf4689fbf3 __clone (libc.so.6 + 0xffbf3)
                
                Stack trace of thread 29480:
                #0  0x00007faf4689505f __poll (libc.so.6 + 0xf505f)
                #1  0x00007faf28785c14 n/a (libpulse.so.0 + 0xdc14)
                #2  0x00007faf28793059 pa_mainloop_poll (libpulse.so.0 + 0x1b059)
                #3  0x00007faf2879d301 pa_mainloop_iterate (libpulse.so.0 + 0x25301)
                #4  0x00007faf2879d3b1 pa_mainloop_run (libpulse.so.0 + 0x253b1)
                #5  0x00007faf2878d61e n/a (libpulse.so.0 + 0x1561e)
                #6  0x00007faf28711d1c n/a (libpulsecommon-13.0.so + 0x1dd1c)
                #7  0x00007faf465bc422 start_thread (libpthread.so.0 + 0x9422)
                #8  0x00007faf4689fbf3 __clone (libc.so.6 + 0xffbf3)
                
[... 136 lines cut ...]
                #17 0x00007faf48971fe0 _ZN9AkElementclERK8AkPacket (libavkys.so.8 + 0x37fe0)
                #18 0x000055ad0445c877 n/a (webcamoid + 0x24877)
                #19 0x000055ad0445e8fd n/a (webcamoid + 0x268fd)
                #20 0x000055ad0445ef63 n/a (webcamoid + 0x26f63)
                #21 0x00007faf46e48762 n/a (libQt5Core.so.5 + 0x2ea762)
                #22 0x000055ad044602e6 n/a (webcamoid + 0x282e6)
                #23 0x00007faf46e48a60 n/a (libQt5Core.so.5 + 0x2eaa60)
                #24 0x00007faf48972116 _ZN9AkElement7oStreamERK8AkPacket (libavkys.so.8 + 0x38116)
                #25 0x00007faf46e48a26 n/a (libQt5Core.so.5 + 0x2eaa26)
                #26 0x00007faf29876a26 _ZN9ScreenDev7oStreamERK8AkPacket (libqtscreen.so + 0xca26)
                #27 0x00007faf29875afb _ZThn16_N12QtConcurrent15RunFunctionTaskIvE3runEv (libqtscreen.so + 0xbafb)
                #28 0x00007faf46c30e92 n/a (libQt5Core.so.5 + 0xd2e92)
                #29 0x00007faf46c2ce0f n/a (libQt5Core.so.5 + 0xcee0f)
                #30 0x00007faf465bc422 start_thread (libpthread.so.0 + 0x9422)
                #31 0x00007faf4689fbf3 __clone (libc.so.6 + 0xffbf3)
                
                Stack trace of thread 29484:
                #0  0x00007faf1c30f6a8 n/a (n/a + 0x0)
                #1  0x00007faf1d3c5026 n/a (swrast_dri.so + 0x637026)
                #2  0x00007faf1d3e5d16 n/a (swrast_dri.so + 0x657d16)
                #3  0x00007faf1d3e5e9e n/a (swrast_dri.so + 0x657e9e)
                #4  0x00007faf1d3c55b8 n/a (swrast_dri.so + 0x6375b8)
                #5  0x00007faf465bc422 start_thread (libpthread.so.0 + 0x9422)
                #6  0x00007faf4689fbf3 __clone (libc.so.6 + 0xffbf3)
```

MADronov, 2020-07-07:
I found somewhat similar issue on Windows (though only app that tries to use Virtual Camera crashes)

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX