webcamoid/webcamoid #333: No options for changing brightness, contrast etc on Mac - detects camera and can get a picture

Humblebum, 2020-07-20:
Using webcamoid on MacBook Air 2013 running High Sierra. Can detect webcam, and can see the stream, but no options available to alter the image. Does this only work for some webcams, or should it work for all. my webcam is a Swift 1.3mp microscope/telescope eyepiece camera.
[image]

hipersayanX, 2020-07-22:
I've tried a lot of times to make camera controls work with every framework in existence:

- **AVFoundation**: The official Apple framework for video capture, does not provides support for UVC controls.
- **CoreMediaIo**: The undocumented Apple framework, I can list the controls but I can't set them.
- **libuvc**: Uses libusb to talk with the camera using the USB protocol, again I can list the controls but I can't set them.

It seems that this is not possible.

8enmann, 2020-11-02:
Why not make a plugin that increases brightness in post-processing?

hipersayanX, 2020-11-05:
@8enmann I had already thought about that, I'll do that, but the results won't be great, hardware controls works modifying the gain of the webcam sensors without loosing any data, so if the scene is dark you can still get a descent image quality increasing the brightness.
if you try to increase the brightness by software you will get a very low quality image because the original data is already lost, and can't be recovered, unless you use some sort of neural network to fix that and simulate the missing data.
An example of that is GNOME Cheese, afair it have (or used to have) software controls and compared to other programs like guvcview or Webcamoid that uses hardware controls, in the first one when you increase the brightness of a picture coming from a dark scene you get a grayish image, in the last two you get perfectly defined image.

8enmann, 2020-11-05:
@hipersayanX what about using gamma correction? I've found this algo to work much better than just modifying the HSL
https://www.pyimagesearch.com/2015/10/05/opencv-gamma-correction/

hipersayanX, 2020-11-05:
@8enmann yes, my idea is to add all those controls (brightness, contrast, saturation, gamma and hue), but again don't expect it to be as good as using hardware controls. Also, I've already wrote [code for that](https://github.com/webcamoid/akvirtualcamera/blob/033c68f3895e532c710b0dfbdf7ee6b13873c849/VCamUtils/src/image/videoframe.h#L655), I just need to make them a plugin.

8enmann, 2020-11-05:
Woohoo, thanks so much!!

dev-broox, 2021-06-18:
@hipersayanX have you checked https://github.com/jtfrey/uvc-util ?
I can change settings on MacOS (10.15.7 Catalina) with webcamoid running using the command line.

>Unlike other (GUI-based) utilities, this code only makes use of the IOKit to walk the USB bus, searching for UVC-compliant devices.

hipersayanX, 2021-06-19:
@dev-broox I tried with [libuvc](https://github.com/libuvc/libuvc) ([here](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VideoCapture/src/libuvc) is the code for the plugin), and it worked well before Catalina, but after Catalina it suddenly started giving me an error when trying to set the controls.
I even tried writing the [code](https://github.com/webcamoid/webcamoid/blob/0aab5f0f22dd6e9967ac5079d1718788a6829fa8/libAvKys/Plugins/VideoCapture/src/avfoundation/src/devicecontrols.cpp#L515) my self, and also tried using [CoreMediaIO](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/cmio/src/capturecmio.mm), and it just did never worked.
Is a total waste of time keep trying, if Apple doesn't not add an official API for the camera UVC controls is futile to keep trying hacks to make the camera controls work.
Also uvc-util does not seem to have any integration with homebrew, pkg-config, or Cmake, it's totally useless for my purposes.

dev-broox, 2021-06-22:
I see. Anyhow, it might be useful for debugging your own implementation. The source is 4 files that could be incorporated (alas, objective-c) or serve as inspiration. e.g. the method for opening the USB channel, which might be related to your problems.

(And yes, developing for Apple is quite a PITA with their half-baked APIs xD)

hipersayanX, 2021-06-22:
@dev-broox 

>The source is 4 files that could be incorporated (alas, objective-c) or serve as inspiration.

I've been reading the code and it does basically the same as my [code](https://github.com/jtfrey/uvc-util/blob/ce78fdc32c5953693b4b9a96b0e1448f61a4f104/src/UVCController.m#L920). Is there something I'm missing here? did you actually tested it can set the controls? :confused: 

> e.g. the method for opening the USB channel, which might be related to your problems.

I think it could be more a problem related to either permissions or entitlements, maybe related to [this](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_device_usb), because from all my tests it can read the controls but isn't  possible to set them.

>(And yes, developing for Apple is quite a PITA with their half-baked APIs xD)

Yes, that and all the walled garden :hankey: :vomiting_face:

dev-broox, 2021-06-30:
> I've been reading the code and it does basically the same as my [code](https://github.com/jtfrey/uvc-util/blob/ce78fdc32c5953693b4b9a96b0e1448f61a4f104/src/UVCController.m#L920). Is there something I'm missing here? did you actually tested it can set the controls? 😕

Yup. I changed gain, brightness, etc. for an external USB camera (the built-in quicktime camera reports no UVC controls).
Open webcamoid, start camera, then in terminal:
```
uvc-util -I 1 -s gain=100
```
The gain (for example) rises up just as expected.

> I think it could be more a problem related to either permissions or entitlements, maybe related to [this](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_device_usb), because from all my tests it can read the controls but isn't possible to set them.

That might be. `uvc-util` will inherit permissions from Terminal.app (which might be quite liberal). That said, is webcamoid sandboxed? I think not. One check would be to execute webcamoid straight from the terminal. If `uvc-util` can change settings, then webcamoid from the terminal should too.

hipersayanX, 2021-06-30:
@dev-broox

>That said, is webcamoid sandboxed? I think not.

And your right.

>One check would be to execute webcamoid straight from the terminal. If uvc-util can change settings, then webcamoid from the terminal should too.

I don't remember if I've tried before, but this could be an option, good tip :+1: 
The code for libuvc is still there, if someone what to try and test.

akumria, 2021-07-28:
Stumbled over this: https://phoboslab.org/log/2009/07/uvc-camera-control-for-mac-os-x

And a few more google searches lead me to your app. Not sure if it heaps but being the control the white-balance hardware a webcam would be great.

hipersayanX, 2021-07-29:
@akumria I remember reading that article and the source code, and took it as reference, and as said it worked fine in Snow Leopard and older. The problem should be something related to entitlements maybe.

hipersayanX, 2022-01-25:
I've added software based camera controls in the latest commit.
I've also tested https://github.com/jtfrey/uvc-util as suggested by @dev-broox, and it doesn't works, it returns:

```
ERROR: unable to write new value to control brightness
```

It can list the controls, but can't modify them, Apple has blocked camera controls somehow.

dev-broox, 2022-02-01:
@hipersayanX well... I just built master in my MBPro, and tried the software controls. 
 
Alas, I can change values with uvc-util for two UVC compatible cameras (a Mokose sony IMX based and  a cheap USB microscope).

What I cannot change, even if uvc-util lists the controls, is the laptop integrated webcam (apple Facetime HD camera) WHEN THE CAMERA IS OPEN. I'll explain:
- For any other USB camera, I can view the camera in either webcamoid or Quicktime player and change any available parameter on another terminal with uvc-util.

- For the *integrated Facetime HD camera* I can change any parameter *when the camera is not open*, and it will have changed when I open the camera in webcamoid and QTP.

- If I try to change a parameter while the *Integrated Facetime HD* camera is open, I get an error like:
```
 uvc-util -I 0 -s brightness=0.5
ERROR:  invalid control name: brightness
```
So It might be camera-dependent, and Apple's Facetime driver does lock the UVC changes.

hipersayanX, 2022-02-01:
@dev-broox I'm doing all my test in VirtualBox, and neither the USB devices or the webcam options are working, they worked on Snow Leopard and older, they stopped working in Catalina, and since the feature is undocumented it doesn't worth the try wasting time with hacks :neutral_face: 
Also, I'll keep giving support to Mac until the last moment, once I can't give support anymore, either other developers step up, or I'll just remove all the code from the tree. Supporting Mac for me is challenging, and I like challenges, but it also adds another layer of problems, and having less problems would be great.

closed 2022-02-15 by hipersayanX