webcamoid/webcamoid #335: Not able to stream video through virtual cam in Firefox using Webcamoid in Xubuntu 20.04

CogitoMax, 2020-07-30:
# Reporting a problem?

## Summary

Not able to stream video through virtual cam in Firefox using Webcamoid in Xubuntu 20.04

## Current Behavior

Video has been added from "Configure Source" in Webcamoid, but when i test in webcam test sites, it is not getting played, but a blue screen of webcamoid is displayed.
[image]

## Expected Behavior

The video i added should be playing instead of blue webcamoid screen.
[image]

## Steps to Reproduce

There is no option to select the video to play. Edit option is greyed-out/disabled in the screen. 
[image]

## Suggestions and tips

General options screen grab is attached, where i think everything is in order.
[image]

## Your Environment

* Webcamoid information: 8.7.1
* Operating System information: Xubuntu 20.04
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Questions and other matters?

How can i stream videoplayback?

CogitoMax, 2020-10-06:
Hello,

I am using the daily build now.
But, the video which i have added is not getting played in output in virtual camera mode.
Here is screenshot of preferences :
[image]

I have tried using webcam tests using various browsers and also in Cheese. 
The output shown is : 
[image]

Whereas, the video i am playing is 
[image]

Why is the program not playing the above video ?

Any how to or help to play video using virtual camera for linux will be much appreciated. 

Thanks!

hipersayanX, 2020-10-11:
The virtual camera seems to be working fine, check in _Video/Outputs_ if the virtual camera is selected.

CogitoMax, 2020-10-13:
In Video/Outputs, Virtual camera is selected. 
[image]
[image]

But, it seems that

CogitoMax, 2020-10-13:
the \

hipersayanX, 2020-10-19:
There is something totally wrong with what you are doing, in the [2nd capture](https://github.com/webcamoid/webcamoid/issues/335#issuecomment-704212821) you show akvcam clearly working under Cheese  and Webcamoid being detecting it in the 3rd and next captures, but the editor is showing that the file can't be found, it that were the case the virtual camera won't be working from very beginning.
Also, why are you using Wine Notepad to edit a file system instead of using a regular GNU/Linux text editor? GNU/Linux noob? :smile:

CogitoMax, 2020-10-21:
Hi!
I am a complete noob to Linux :) and i am using Xubuntu xfce environment (20.04). I didn't modify anything. I guess the notepad looks like this by default in xfce.

Will try to reinstall akvcam from the scratch.

I tried uninstalling akvcam as per the commands stated [here](https://github.com/webcamoid/akvcam/wiki/Build-and-install)
But, it didn't remove the akvcam folder inside /etc/

Just an update though, when i changed the Virtual Camera Driver to v4l2loopback, it works fine in Firefox now, but not in Chrome.

hipersayanX, 2020-10-21:
>I am a complete noob to Linux :)

Ha, I knew it! also welcome to the land :smile: 

>and i am using Xubuntu xfce environment (20.04). I didn't modify anything. I guess the notepad looks like this by default in xfce.

You probably installed Wine recently and the installer configured Wine notepad as the default editor, I don't know Xfce but check in the file properties (or similar option) and select gedit, leafpad or any other more suitable editor.

>Will try to reinstall akvcam from the scratch.

No, is not necessary. The "could not be found " message above is because the file has 600 permissions, that is the file is only readable by the root user (thing that have not much sense anyways, I've updated the wiki), you can read the file with:

 ```
sudo cat /etc/akvcam/config.ini
 ```

>I tried uninstalling akvcam as per the commands stated here
>But, it didn't remove the akvcam folder inside /etc/

The file is created manually, and you must remove manually. .

>Just an update though, when i changed the Virtual Camera Driver to v4l2loopback, it works fine in Firefox now, but not in Chrome.

Because v4l2loopback identifies it self as a M2M device and many programs reject those devices, akvcam instead identifies as a capture device, check the [compatibility list](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support#gnulinux).

hipersayanX, 2020-10-26:
Anyway, seems akvcam is broken again with newer kernels :angry:

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX