webcamoid/webcamoid #337: Segmentation fault: warnings about depreciated properties and error about anchors on an item managed by a layout

narilth, 2020-08-04:
# Reporting a problem?

## Summary

`webcamoid` segfaults after making window with warnings about depreciated properties in QML Connections and an error about undefined behavior about anchors on an item managed by a layout in QML Label.

## Current Behavior

When running `webcamoid`, the program opens the Webcamoid window. Then, the webcam light turns on.

After about a second, the window closes. The command line reports that the program had a segmentation fault. It also displays warnings about depreciated properties in QML Connections and an error about undefined behavior about anchors on an item managed by a layout in QML Label. The log is in the "Your Environment" section of this issue.

## Expected Behavior

The window should stay open and should display the video produced by the webcam.

## Steps to Reproduce

1.  Run `webcamoid` in a terminal.
2.  Wait a few seconds.
3.  The program has now crashed.

## Suggestions and tips

I would remove the anchors on the QML Label and do what the QT warnings tell me to do.

## Your Environment

* Webcamoid information: `webcamoid --version: Webcamoid 8.7.1`
* Operating System information: Arch Linux ~~(numeric version and codename if applied)~~ x64

### Any other useful information:

The logs: 
```
~  $ webcamoid
qrc:/Webcamoid/share/qml/RecordConfig.qml:82:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/RecordConfig.qml:74:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/Webcamoid/share/qml/EffectConfig.qml:58:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/EffectConfig.qml:53:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/AudioInfo.qml:184:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/MediaConfig.qml:46:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/MediaConfig.qml:40:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/OutputConfig.qml:35:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/VirtualCamera/share/qml/main.qml:47:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/RecordBar.qml:87:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/EffectBar.qml:93:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/AudioConfig.qml:78:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/MediaBar.qml:53:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/main.qml:127:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/main.qml:122:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/main.qml:117:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/Webcamoid/share/qml/main.qml:102:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/MultiSink/share/qml/main.qml:143:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
zsh: segmentation fault (core dumped)  webcamoid
```
The window before the program crashes:[image]

hipersayanX, 2020-08-10:
Try installing [webcamoid-git](https://aur.archlinux.org/packages/webcamoid-git) instead. The warnings will be fixed later, it does not affects the funtionality.

narilth, 2020-08-11:
webcamoid-git works, thank you.

closed 2020-08-11 by narilth