webcamoid/webcamoid #339: Not being able to Run in Qt Creator 4.12.4

talobin, 2020-08-13:
## Summary

I am having a problem running the app after finished building in QT
## Current Behavior

I have finished the steps listed here: https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows
I got my by build step to finish without error.
My build configurations are : **Desktop Qt 5.13.2 MSVC2017 64 Bit -> Debug -> StandAlone**
Now I cannot get it to run.  Here is the error from "Application Output" of QT:

21:42:21: Starting C:\Users\vodin\Documents\DJI\RnD\build-Webcamoid-Desktop_Qt_5_13_2_MSVC2017_64bit-Debug\StandAlone\debug\Qt5.13.2\cl\x86_64\bin\webcamoid.exe -r -p C:\Users\vodin\Documents\DJI\RnD\webcamoid\libAvKys\Plugins --vcam C:\Users\vodin\Documents\DJI\RnD\webcamoid\libAvKys\Plugins\VirtualCamera\src\dshow\VirtualCamera
...
QML debugging is enabled. Only use this in a safe environment.
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
Error 20 (this feature has not been implemented yet) in function AVolute::GetProductInfoT::<lambda_3920e95365a48b95dd51020986e9e351>::operator ()
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:430 Type VideoEffectsPanel unavailable
qrc:/Webcamoid/share/qml/VideoEffectsPanel.qml:48 Type VideoEffectsList unavailable
qrc:/Webcamoid/share/qml/VideoEffectsList.qml:24 Type ScrollView unavailable
qrc:/Webcamoid/share/themes/WebcamoidTheme/ScrollView.qml:38 Non-existent attached object

## Expected Behavior

The app launches

## Steps to Reproduce

1. Rebuild.
2. Run

## Your Environment

* Webcamoid information: Master Branch/ Latest commit
* Operating System information: Windows 10 Pro
* QT Creator 4.12.4

Thank you!
If anyone has successfully built and run in Windows, please tell me your QT version and build config, I would love to try that out.

talobin, 2020-08-17:
I have tried to change from  
Desktop Qt 5.13.2 MSVC2017 64 Bit -> Debug -> StandAlone
to
Desktop Qt 5.**15.**0 MSVC2019 64 Bit -> Debug -> StandAlone (by installing QT 5.14.2)

and 

1. If using 8.7.1 TAG, it won't compile because of many compilation errors:
[image]

2. If using current Mater, it will compile, but when I run, the app looks very simplified, nothing like the release app. Also,
most of the features do not work:
[image]

Thank you.

talobin, 2020-08-17:
I can develop on any platform setup.  If there is someone out there that is actively working on this, please let me know your platform and all the detail and I would be gladly to set up my environment to be the same just to get my project going.
Thank you so much.

rayx8426, 2020-09-29:
In order to build tag 8.7.1 version, you need to
1. add `#include <string>` to `persistpropertybag.h`
2. modify line 1085~1086 in `pin.cpp` to
```
static std::uniform_int_distribution<int> distribution((std::numeric_limits<uint8_t>::min)(),
                                                           (std::numeric_limits<uint8_t>::max)());
```

also set additional argument to [Run] `-q C:[path]\webcamoid\libAvKys` 

wiki for tag 8.7.1:
https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows/c8f1b38b14811985673211afdd948ed039ba7a26

hipersayanX, 2020-10-03:
I've updated [the wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows), since Qt is not releasing pre-made binaries anymore and Zeranoe FFmpeg builds are over, I've switched the development to MSYS2, Webcamoid won't support MSVC build until a valid development environment appears.

hipersayanX, 2022-02-15:
The minimum required now is Qt 5.15, closing.

closed 2022-02-15 by hipersayanX