webcamoid/webcamoid #341: Settings persisting after uninstall

hysteria357, 2020-08-25:
## Summary

After uninstalling, the settings I used before uninstalling stayed on the camera after uninstall, and persist even through a reinstall of the camera drivers. 

Windows 10 64-bit.
Webcamoid 8.7.1

I read issue #178 and the suggestions there did not work.  More detail below

## Current Behavior

I uninstalled webcamoid to go back to the default camera software which came with the webcam (Razer Synapse).  When I uninstalled the program, all the configurations I kept in webcamoid persisted on the camera regardless of the settings I used in the razer program.  I tried uninstalling the camera drivers and re-installed, and every camera application still pulls the webcamoid settings.  This is extremely confusing as the program is completely uninstalled.  Loading up, programs still read "USB Video Device" instead of the camera name "Razer Kiyo".  I do not see a virtual webcam service running, I do not see any registry keys, and the batch file described in issue #178 did not work because there is no virtual camera folder under c:/program files.

## Steps to Reproduce
Uninstall webcamoid, open any program which uses the webcam.  

Thank you.  The software worked well when I was in a static environment but now I am in an apartment with lots of lighting changes due to a lot of windows and I need auto adjustment.

robbak, 2020-08-28:
This sounds like issue #148 . Check the solutions on that issue, or install the nightly, then uninstall it, which should fix the problem.

hipersayanX, 2020-08-30:
Same bug as #148, closing.

closed 2020-08-30 by hipersayanX