webcamoid/webcamoid #343: Can Webcamoid rotate a video feed?

zylstra, 2020-08-30:
Can Webcamoid rotate a video feed?

hipersayanX, 2020-09-03:
There's no specific plugin for that, but you can use **Matrix Transform** to rotate the source, for example the following matrix rotate the video 45°:

```
0.7071 -0.7071
0.7071  0.7071
```

https://en.wikipedia.org/wiki/Rotation_matrix

dchmelik, 2022-11-09:
> There's no specific plugin for that, but you can use **Matrix Transform** to rotate the source, for example the following matrix rotate the video 45°:

I need this to mirror-flip photo on Webcamoid 8.8 (9 won't even save photo) but there's no 'save/apply' button(s) for rotation matrix effect?

hipersayanX, 2022-11-09:
Now there is a Rotate plugin in the [video-converter branch](https://github.com/webcamoid/webcamoid/tree/video-converter), I'll release it with 9.1.0.
@dchmelik there is no need for an apply button, the changes are applied intermediately.

hipersayanX, 2023-06-10:
Feature added in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX

stav-ros, 2024-02-21:
> Feature added in Webcamoid 9.1.0.

Hi, there is a rotate function, but is there a mirror-flip function in the works?