webcamoid/webcamoid #347: on macos, there is alyways a crash when I call CMIOObjectAddPropertyListener to listen device change

zzk167, 2020-09-06:
when I call CMIOObjectAddPropertyListener(kCMIOObjectSystemObject, inAddress, inListener, inClientData) to listen device change event, a carsh alway happens.

const CMIOObjectPropertyAddress propertyAddress = {
kCMIOHardwarePropertyDevices,
kCMIOObjectPropertyScopeGlobal,
kCMIOObjectPropertyElementMaster
};

the dump in my app:
perating system: Mac OS X
                  10.15.6 19G73

CPU: amd64
     family 6 model 142 stepping 9
     4 CPUs

Crash reason:  EXC_BAD_ACCESS / KERN_INVALID_ADDRESS
Crash address: 0x0
Crash time:    1599207333

Thread 0 (crashed)
  0  libsystem_platform.dylib + 0xe52
      r10 = 0x00007f9296679c00      r11 = 0x00007fff6460a3de
      r12 = 0x0000000000000000      r13 = 0x00007f929667df00
      r14 = 0x00007ffeeb5d5860      r15 = 0x00007f929661a3d0
       r8 = 0x0000000000000004       r9 = 0x0000000000000004
      rax = 0x0000000000000000      rbp = 0x00007ffeeb5d57a0
      rbx = 0x0000000000000000      rcx = 0x0000000000000000
      rdi = 0x0000000000000000      rdx = 0x0000000000000000
      rip = 0x00007fff674fde52      rsi = 0x0000000000000000
      rsp = 0x00007ffeeb5d57a0
     Found by: given as instruction pointer in context
  1  libc++.1.dylib + 0x383f2
      rbp = 0x00007ffeeb5d57c0      rip = 0x00007fff6460a3f3
      rsp = 0x00007ffeeb5d57b0
     Found by: previous frame's frame pointer
  2  AkVirtualCamera + 0x4540
      rbp = 0x00007ffeeb5d5a00      rip = 0x0000000109ceb541
      rsp = 0x00007ffeeb5d57d0
     Found by: previous frame's frame pointer
  3  AkVirtualCamera + 0x633b
      rbp = 0x00007ffeeb5d5a20      rip = 0x0000000109ced33c
      rsp = 0x00007ffeeb5d5a10
     Found by: previous frame's frame pointer
  4  AkVirtualCamera + 0x4324
      rbp = 0x00007ffeeb5d5a40      rip = 0x0000000109ceb325
      rsp = 0x00007ffeeb5d5a30
     Found by: previous frame's frame pointer
  5  CoreFoundation + 0xfd75c
      rbp = 0x00007ffeeb5d5a50      rip = 0x00007fff2d31675d
      rsp = 0x00007ffeeb5d5a50
     Found by: previous frame's frame pointer
  6  CoreFoundation + 0xfd5c6
      rbp = 0x00007ffeeb5d5a90      rip = 0x00007fff2d3165c7
      rsp = 0x00007ffeeb5d5a60
     Found by: previous frame's frame pointer
  7  CoreMediaIO + 0x111f8
      rbp = 0x00007ffeeb5d5b90      rip = 0x00007fff2e5261f9
      rsp = 0x00007ffeeb5d5aa0
     Found by: previous frame's frame pointer
  8  CoreMediaIO + 0x458f
      rbp = 0x00007ffeeb5d5c50      rip = 0x00007fff2e519590
      rsp = 0x00007ffeeb5d5ba0
     Found by: previous frame's frame pointer
  9  CoreMediaIO + 0x4113
      rbp = 0x00007ffeeb5d5c80      rip = 0x00007fff2e519114
      rsp = 0x00007ffeeb5d5c60
     Found by: previous frame's frame pointer
 10  CoreMediaIO + 0x3f7e
      rbp = 0x00007ffeeb5d5c90      rip = 0x00007fff2e518f7f
[... 140 lines cut ...]
VM Region Summary:
ReadOnly portion of Libraries: Total=400.5M resident=0K(0%) swapped_out_or_unallocated=400.5M(100%)
Writable regions: Total=81.6M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=81.6M(100%)
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
Dispatch continuations            24.0M        1 
Kernel Alloc Once                    8K        1 
MALLOC                            48.1M       18 
MALLOC guard page                   16K        4 
STACK GUARD                       56.0M        3 
Stack                             9232K        3 
__DATA                            2049K       46 
__DATA_CONST                        20K        1 
__LINKEDIT                       388.8M        3 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                            11.7M       45 
__UNICODE                          564K        1 
mapped file                         32K        1 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                            574.6M      136

zzk167, 2020-10-15:
is there some updates?

hipersayanX, 2020-10-15:
@zzk167 I've splitted the virtual camera into it's [own project](https://github.com/webcamoid/akvirtualcamera), beside that I haven't checked this issue yet.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX