webcamoid/webcamoid #348: Infrared camera being used

Trev57, 2020-09-09:
## Summary

My webcam - WYZEC2 - is a security cam with some Webcam firmware pasted into it, it works fine in every other application, OBS, XSplit, Chrome, and Zoom, all work with it. However for some reason the webcam goes into night vision mode for this and everything is green and grainy.

## Current Behavior

The webcam input is in infrared mode so it is all green and grainy

## Expected Behavior

It should not be in infrared mode
[image]

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Get a WYZEC2 camera, or possibly others
2.  Download and install the firmware correctly (its official firmware on their website)
3.  Start webcamoid

## Suggestions and tips

There is a very very short time, after starting up webcamoid for the first time, that the camera shows a normal image, then it suddenly switches to night mode. Something in webcamoid is taking what brightness the camera is giving it, and then setting the webcam into night mode too early, or at too high a brightness. 
And yes the video is upside down and I am trying to use this to flip it for a virtual webcam

## Your Environment

* Webcamoid information: 8.7.1 x64 
* Operating System information: Windows 10 
* Any other useful information: its a USB A to A connection

hipersayanX, 2020-09-12:
What formats does it provides? Does it show any control related to night mode? Can you give me a screenshot of this part?

[image]

Trev57, 2020-09-12:
Video formats are MJPG, and YUY2
[image]

hipersayanX, 2020-09-12:
It seems to be detected as a normal webcam, and I've found [this](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/infrared-stream-support-in-uvc), because the format for normal and IR capture are the same, Webcamoid is just taking the last one and ignoring the other, a solution would be reading those flags and marking that format as something different to the normal format. Will check.

Trev57, 2020-09-13:
update on the issue, the webcam will continue to do this until webcamoid is completely uninstalled using the instructions at #178

setswservices, 2020-12-13:
I just installed the 64-bit version on a Windows 10 Pro system, build 19041.685, version 2004. Nothing happens when launched, though webcamoid shows up in the Task Manager, consuming 0% CPU time. I tried building it and launching and same thing.  Would you please tell me what version of Windows you have?