webcamoid/webcamoid #349: Headless / Daemon mode?

ElDavoo, 2020-09-12:
I'm using webcamoid to horizontally flip a webcam. However, since this is a permanent setup with an always turned-on device, I'd like webcamoid to start up automatically when windows starts and as a daemon- no interface should be present.

hipersayanX, 2020-09-12:
So basically, start minimized?

ElDavoo, 2020-09-12:
Kind of. The most simple implementation would be a tray icon and an option
to make the windows disappear when minimised + auto boot in windows + auto
boot minimised.

A fully fledged implementation would be a configurable Daemon/service and a
front-end for that Daemon but that implies a full code rewrite lol

hipersayanX, 2020-09-12:
The first one is the easier, the second one will be a design hell and and totally out of the project goals.