webcamoid/webcamoid #351: Not Opening On Windows 10

JamesHezzy, 2020-09-18:
So I've tried multiple times to get this to open, including resetting my configs multiple times. I've tried both portable and installed versions. 

When I open/run the program I get the icon pop up in my taskbar but the program itself doesn't seem to fully start. I'm not sure if I'm missing some sort of prerequisite to run the program, I don't think I am. The program runs fine on my laptop, just not on my desktop. 

I'm on windows 10 ver1909

hipersayanX, 2020-09-18:
Try [resetting the configurations](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh).

JamesHezzy, 2020-09-19:
> Try [resetting the configurations](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh).

JamesHezzy, 2020-09-19:
As I said in my initial post, that is something I have tried many times. I deleted the registry entry and installed, no luck. I deleted and tried the portable, again no luck.

Valinwolf, 2020-10-31:
I can confirm this issue exists on Windows 10 Pro version 20H2, build 19042.610. I also just installed it so there are no configs to reset.

setswservices, 2020-12-13:
I just installed the 64-bit version on a Windows 10 Pro system, build 19041.685, version 2004.  Nothing happens when launched, though webcamoid shows up in the Task Manager, consuming 0% CPU time.  I tried building it and launching and same thing.

setswservices, 2020-12-13:
> I can confirm this issue exists on Windows 10 Pro version 20H2, build 19042.610. I also just installed it so there are no configs to reset.
> 
> P.S.: The process opens and runs in the background but no window shows.

setswservices, 2020-12-13:
> I can confirm this issue exists on Windows 10 Pro version 20H2, build 19042.610. I also just installed it so there are no configs to reset.
> 
> P.S.: The process opens and runs in the background but no window shows.

I can confirm this.

setswservices, 2020-12-13:
I found that if I wait 10 minutes or more (!), Webcamoid appears, and is visible as an App in Task Manager.  Before that happens, for many minutes, it appears as a background process.

agp8x, 2021-01-13:
For me, the Window was on a disconnected monitor. #117 seems to be related to that problem

hipersayanX, 2021-01-25:
@agp8x that happened in Webcamoid 8.7, right? did you tried the daily build?

agp8x, 2021-01-25:
Yes, that was 8.7.1.
When dragging the window from the reconnected screen, it shrinked itself until it vanished.

With daily, I could not find a window and switched to OBS

adrianhoehne, 2021-05-29:
I can confirm with 8.8.0. Windows error reporting pops up for a second after the application closed. At startup the application window appear for a few seconds then it crash:

Windows Report:
Name der fehlerhaften Anwendung: webcamoid.exe, Version: 8.8.0.0, Zeitstempel: 0x00000000
Name des fehlerhaften Moduls: swscale-5.dll, Version: 5.7.100.0, Zeitstempel: 0x00000000
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0000000000066e81
ID des fehlerhaften Prozesses: 0x337c
Startzeit der fehlerhaften Anwendung: 0x01d754933ff6f9d6
Pfad der fehlerhaften Anwendung: C:\Program Files\webcamoid\bin\webcamoid.exe
Pfad des fehlerhaften Moduls: C:\Program Files\webcamoid\bin\swscale-5.dll
Berichtskennung: 5cbc3c72-8cbb-4f00-8a28-3ecd6d61647b

hipersayanX, 2021-05-30:
@adrianhoehne its probably some bug when scaling the frames. You can try disconnecting the camera ant then, go to Preferences > General Options > and disable Play webcam on start. It would be useful if you can give me a list of resolutions supported by your webcam.

adrianhoehne, 2021-05-31:
It was at 1920x1080 and 640x480 and supported MJPG with 60fps and YUV with 30fps. 
I already did this "disabling Play webcam on start" it helped but anyway, I send back the camera, it was not working for my purposes. Sorry.

11cs103, 2021-10-31:
The default video format on webcamoid is set as "MJPG" which when changed to "YUY2" format the app works fine but not smooth as the camera functions without webcamoid! 

Changing the video format to "MJPG" format again causes the app to hang and crash. When I tried restarting the pc there was a notification about "the dynamic resolution" being carried out in the background. This only happened after installing webcamoid. 

I hope this bit information from my experience helps whomever looking to fix this bug!

MintMain21, 2022-02-05:
Discovered this software on Linux Mint 20.3. Installed it on Windows 10 because I loved it so much, and after installing it likewise crashes on startup every time. It's a shame, because this gives more native Windows applications of similar purpose a run for there money.

I mean, the Xbox Game Bar is hardly an acceptable tool for taking screenshots.

vctls, 2022-02-07:
Here too, app crashes as soon as I set video format to MJPG.
YUY2 works fine but is way less performant than what MJPG offered when it worked.
YUY2 appears to support a wider range of resolutions, up to 2304x1536, which I'm not event sure this camera is supposed to support, but the framerates are way lower. 5fps in 1080p instead of 30 for MJPG.

Window 10 21H1 build 19043.1503
Webcamoid 8.8.0
Logitech HD Pro C920

hipersayanX, 2022-02-13:
@MintMain21 @vctls just try the daily build, or wait 2 weeks more for 9.0.0 release, this bug was fixed already.

vctls, 2022-02-13:
Can confirm it works on my end, thank you @hipersayanX !

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX

tomdevlive, 2023-03-27:
I have literally never gotten this app to work. Been trying for like 2 years. It just sits there saying Not Responding. I'm on Windows 10. Tried both the portable and exe. Willing to share whatever debug log I need to. Would really like to use this app.

MintMain21, 2023-03-27:
@tomdevlive 

Have you tried installing the latest version?

tomdevlive, 2023-03-28:
@MintMain21 yes.

hipersayanX, 2023-03-29:
@tomdevlive Does your computer have something special? Did it in some moment worked or absolutely never?

hipersayanX, 2023-03-29:
@tomdevlive  you can try try running Webcamoid with ```webcamoid --log-file log.txt``` but I don't think the out will be much helpful, also in the ```lib/avkys``` you have all the plugins, try moving all the plugins with the following pattern outside Webcamoid folder:

```
libAudioDevice_*
libDesktopCapture_*
libMultiSink_*
libMultiSrc_*
libVideoCapture_*
libVirtualCamera_*
```

Check if you can launch Webcamoid, if it does, try returning them one by one until it hung again, inform me which one was the problematic one.