webcamoid/webcamoid #352: Color Accuracy Issues with Virtual Webcam (8.7.1)

MikeDFWM, 2020-09-22:
When using vMix External as my webcam in webcamoid, this is the feed I'm sending to the program:

[image]

This is how it looks in webcamoid:

[image]

The problem is that, when I select my virtual webcam as my vMix Camera Input (yes, I have a reason for doing this & it's actually my sole reason for seeking out & using this program), this is the feed as it appears in vMix:

[image]

You'll notice, in the yellows especially, that the color accuracy is off compared to what's sent to, and received by, the program. The yellow is dirtier, the magenta is more purple, while the blues are somehow brighter.

Here's a side-by-side shot of the vertical yellow bars, to make it clearer:

[image]

The left is the yellow as it appears on the webcamoid screen. The right is the yellow as it appears in the vMix feed of the webcamoid virtual webcam.

To see if it's a vMix issue, I opened the virtual webcam in Skype. As you can see (though the image is mirrored), the same color differences are present:

[image]

I'll point out again that the vMix feed & webcamoid preview are identical (at least to my eyes, which is good enough). The virtual webcam feed of that image, in the 2 applications I tested it on (and I'd assume more) , however, represents the colors differently than both.

MikeDFWM, 2020-09-22:
I forgot to mention, I'm on Windows 10 (Version 2004). The virtual webcam is using the AkVirtualCamera Driver, which is the only option in settings.

The version of Webcamoid that I'm using is "webcamoid-portable-8.7.1-win64"

hipersayanX, 2020-09-22:
The colors are different because you are sending an RGB image (either 24 or 32 bpp), and in the other side the frame is converted to a YUV format (either 12 or 16 bpp) which has lower color resolution and therefore degrading the picture quality, if you want the same color quality in the target program you must set a RGB format in there.

MikeDFWM, 2020-09-22:
There's no setting in Skype that I can find that allows that level of finesse to my knowledge. And I don't run into this problem with any other virtual webcam, including going directly to Skype from vMix itself. 

So while I have no reason to doubt you & every reason to believe that your diagnosis of the issue is correct, it's not something that's easily (or even necessarily possibly) remedied while simultaneously being a behavior that's different from every other virtual webcam I've used while testing within the same software.

I'd be happy to post images using other virtual webcams showing that this issue is unique to this program's output, at least among the ones I've used. OBS Virtual Cam, ManyCam, and vMix all behave as expected.

So, as I said, while I have no reason to doubt you, saying the program is in the receiving program strikes me as a bit oversimplifed when every other virtual webcam I've used, in those same receiving programs, provides an image where, if color accuracy is off, it's not off enough for me to notice, while this one was readily apparent.

hipersayanX, 2020-09-22:
Thee color miss matching is not a problem because that's how color space conversion works.

>There's no setting in Skype that I can find that allows that level of finesse to my knowledge.

Webcamoid 9 will have more [options for controlling the formats and resolutions offered by the virtual camera](https://github.com/webcamoid/webcamoid/issues/216#issuecomment-633117608), so even if Skype does not have such options you will able to control such features.

MikeDFWM, 2020-09-22:
So it's not a problem, but you're going to solve it anyway?

Seems like an odd reaction to something that's not a problem. Especially when it's something that isn't present in other similar software under the same conditions.

But thank you.

closed 2020-09-22 by MikeDFWM

hipersayanX, 2020-09-22:
>So it's not a problem,

No, is not.

>but you're going to solve it anyway?

Yes, read the previous comment again.