webcamoid/webcamoid #353: Failing to Launch - Linux

noirtier1, 2020-09-23:
Hi, folks;
I have tried a few times to install this program on Linux Mint 64-bit. The install seems to go through just fine, but once it's finished, launching the program just causes it to think for a while and give up. I tried running from the command line, and got the following:

> @Socrates:~$ webcamoid
> qt5ct: using qt5ct plugin
> qt5ct: D-Bus system tray: no
> QGLXContext: Failed to create dummy context
> qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
> qt5ct: palette support is disabled
> Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
> Aborted (core dumped)

I tried a number of different ways of searching, but so far haven been unable to find anything that seems helpful. If anyone has any suggestions, I would appreciate it. I have been struggling to try and find a way to make camera adjustments - light/dark, that sort of thing - for Zoom meetings. No luck so far.

Thanks in advance!

hipersayanX, 2020-09-23:
Uninstall anything related to qt5ct plugin and try launching again, qt5ct is known to be problematic.

noirtier1, 2020-09-23:
Thank you. I tried that, reinstalled both Webcamoid and the plugins, and saw the same behaviour. I had uninstalled quite a few QT5 entries from the Software Manager, but after a little digging found a suggestion for purging everything from the command line. 

> $ sudo apt-get purge --auto-remove qt5-default 
[https://www.thelinuxfaq.com/ubuntu/ubuntu-17-04-zesty-zapus/qt5-default?type=uninstall]

At this point, I still saw the same issue. Another discussion mentioned that the versions in Software Manager and on your site were different; After downloading the install file and running that, it looks like I've gotten over my initial speedbump.

Much appreciated.

closed 2020-09-23 by noirtier1