webcamoid/webcamoid #354: Adding Virtual Webcam does nothing (MacOS)

Swicegood, 2020-09-24:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

Swicegood, 2020-09-24:
Add Output dialog comes up. Add virtual cam dialog comes up, but leaving defaults and choosing "OK" does not add and entry to the list of output. Nor does a virtual webcam show up (at least in Zoom)

Expected behavior:
Should work just like it does in Windows. A virtual webcam entry shows up in outputs. And then Zoom will allow me to choose that as a camera.

Steps:
Click "Video"
Click "Outputs"
Click "Add output"
Click "Ok' on dialog that comes up.

wecamoid version: development as of 9/24/2020
OS: MacOS 10.15.3

thiagoeh, 2020-09-25:
I'm in a similar situation.
Looks like the VirtualCam feature don't recognize the real webcam, and the list of input devices become empty.
Thanks to that, the VirtualCam is not functional, as there is no source for it.

hipersayanX, 2020-09-25:
The virtual camera is now splitted to [it's own project](https://github.com/webcamoid/akvirtualcamera) you can install it from [daily build](https://bintray.com/webcamoid/webcamoid/akvirtualcamera/daily/), there is not a stable release yet.
The virtual camera **is not** compatible with Webcamoid 8.7 and older.

closed 2021-02-16 by hipersayanX