webcamoid/webcamoid #355: Webcamoid not working after Droidcam

vlig, 2020-09-25:
Hello! After playing with Droidcam, on the next start, Webcamoid doesn't do anything. Buttons are able to be pressed, but no response at all. Deleting ~/.config/Webcamoid isn't helpful. OBS works smooth, showing my webcam well.
And after close, there is a process in memory with ~12% of CPU using.
[image]
[image]

hipersayanX, 2020-10-03:
This issue is probably related to #259 and was solved in the daily build.

ppcutter, 2021-03-21:
Same issue here with Ubuntu 20.04.2 LTS linux x86_64 Webcamoid 8.8.0 and Droidcam. With Webcamoid 8.7.1 everything works fine.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX