webcamoid/webcamoid #356: Program crashes with a segmentation fault

Gedoix, 2020-10-02:
Whenever I open webcamoid, the program crashes after a couple of seconds.

This, however, doesn't happen when I open the program at the same time as being inside a zoom conference, for some reason. Probably having to do with camera usage.

When I do that, though, the integrated camera doesn't display anything in webcamoid, and the pictures and video recordings don't do anything with it either. All features work on screen0, but none on the camera.

As soon as I exit the zoom meeting, the program crashes again.

## Current Behavior

When run on terminal on its own, exits with:

```{bash}
Segmentation fault (core dumped)
```

When in `gdb`:

```{bash}
Thread 21 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff6240a640 (LWP 20732)]
0x00007fffe8a0c852 in ?? () from /usr/lib/libswscale.so.5
```

When run in `gdb` in a zoom meeting, and set to the integrated camera (independently of the video format value I choose) it prints:

```{bash}
qrc:/VideoCapture/share/qml/main.qml:272:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
libv4l2: error setting pixformat: Device or resource busy
VideoCapture: Can't set format: "MJPG"
qrc:/Webcamoid/share/qml/OptionList.qml:53: TypeError: Cannot read property 'right' of null
qrc:/Webcamoid/share/qml/OptionList.qml:54: TypeError: Cannot read property 'left' of null
```

And when I close zoom, after a few seconds:

```{bash}
--Type <RET> for more, q to quit, c to continue without paging--
qrc:/Webcamoid/share/qml/OptionList.qml:54: TypeError: Cannot read property 'left' of null
Thread 71 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff577fe640 (LWP 24621)
0x00007fffe8a0c852 in ?? () from /usr/lib/libswscale.so.5
```

## Expected Behavior

It shouldn't crash, and when not in a zoom meeting it should show the camera video.

## Steps to Reproduce

Open webcamoid.

For the zoom test I just start a call with myself, turn on the camera, and then run webcamoid.

## Environment

* Webcamoid information: Installed from the AUR using `yay`: [https://aur.archlinux.org/packages/webcamoid/](https://aur.archlinux.org/packages/webcamoid/)
* Operating System information: Manjaro linux, on kernel 5.4, fully updated, `screenfetch` outputs:

```{bash}
 OS: Manjaro 20.1.1 Mikah
 Kernel: x86_64 Linux 5.4.67-1-MANJARO
 Shell: bash 5.0.18
 DE: KDE 5.74.0 / Plasma 5.19.5
 WM: KWin
 CPU: Intel Core i5-7300U @ 4x 3.5GHz [37.0°C]
 GPU: Intel Corporation HD Graphics 620 (rev 02)
```

Gedoix, 2020-10-02:
Ah, `webcamoid --version` outputs:

```{bash}
Webcamoid 8.7.1
```

hipersayanX, 2020-10-03:
Did you tried if this happens in the daily build too?

Gedoix, 2020-10-05:
Sorry, I haven't

andreesteve, 2020-10-05:
Hi,

I had a repro on the same version.

I compiled the sources from commit g7f1cb129 and the SEGFAULT does not happen anymore.

andrejpodzimek, 2020-12-21:
I'm getting segfaults with release 8.7.1 as well as with the head version. In `dmesg` they look like this:

```
[ 3741.535325] Thread (pooled)[54629]: segfault at 20 ip 00007f5ec2e98fed sp 00007f5dafffe8e0 error 4 in libgstreamer-1.0.so.0.1802.0[7f5ec2e70000+b5000]
```

I've tried `webcamoid --no-load='/usr/lib/avkys/submodules/ACapsConvert/libgstreamer.so;/usr/lib/avkys/submodules/MultiSink/libgstreamer.so;/usr/lib/avkys/submodules/MultiSrc/libgstreamer.so;/usr/lib/avkys/submodules/VideoCapture/libgstreamer.so'`, but to no avail; it still segfaults. (Well, it's more than likely that I just used `--no-load` incorrectly.)

hipersayanX, 2020-12-28:
@andrejpodzimek blacklisting submodules is not possible yet, try deleting those files:

```
sudo rm -vf \
    /usr/lib/avkys/submodules/ACapsConvert/libgstreamer.so \
    /usr/lib/avkys/submodules/MultiSink/libgstreamer.so \
    /usr/lib/avkys/submodules/MultiSrc/libgstreamer.so \
    /usr/lib/avkys/submodules/VideoCapture/libgstreamer.so
```

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX